micropython/micropython #9076: BLE UART REPL example not working on esp32

bx5974, 2022-08-20:
Hello, 

Im trying to get the ble UART repl example to run on esp32, on micropython 1.19.1 

the below example doesn't seem to work, 
```
/pyboard> ls
ble_advertising.py       ble_uart_peripheral.py  
ble_simple_peripheral.py ble_uart_repl.py        
/pyboard>

```
on my repl, 

```
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
>>> exec(open('ble_uart_repl.py').read())
>>>

```
I dont see any BLE repl starting to advertise - please advise 

[ble_uart_repl.py](https://github.com/micropython/micropython/blob/v1.19.1/examples/bluetooth/ble_uart_repl.py
)

jimmo, 2022-08-21:
See ble_uart_repl.py - it doesn't do anything automatically other than define functions and classes. You need to call start.

> exec(open('ble_uart_repl.py').read())

Rather than doing this you can just `import ble_uart_repl`, and then do `ble_uart_repl.start()`
Reading the whole file into RAM is quite expensive.

bx5974, 2022-08-21:
@jimmo thank you so much, appreciate your quick response

bx5974, 2022-08-21:
closing this, thanks

closed 2022-08-21 by bx5974