micropython/micropython #9078: Watchdog triggers before timeout on RP2 (specifically WS5100S-EVB-PICO)

AndyThirtover, 2022-08-21:
Example

 `from machine import WDT`
`wdt=WDT(timeout=600000)`  expect a 10 minute timeout

Wait a few seconds and the system will reboot.

`MicroPython v1.19.1 on 2022-06-18; W5100S-EVB-Pico with RP2040`

Discovered when testing the following:

```
from machine import WDT, reset
import uasyncio as asyncio
import network

"""
   WATCHDOG -- Reboot system if Network not seen for 600 seconds
   or any other system crash
"""

mon_period = const(600000)
dog_period = const(600000+1000)

wdt = WDT(timeout=dog_period)

async def monitor():
	while True:
		wdt.feed()
		await asyncio.sleep_ms(mon_period) #  i.e. shorter than dog period
		print ('Watchdog period COMPLETE')
```

Regards

Andy

robert-hh, 2022-08-21:
From the data sheet:

> The watchdog (Section 4.7) has a 24-bit counter, that decrements every tick, starting from a user defined value set in LOAD register. There is a logic error which means the counter is decremented twice per tick, instead of once per tick. In a recommended setup where the tick occurs at 1μs intervals, this halves the maximum time between resetting the watchdog counter from ~16.7 seconds to ~8.3 seconds.

AndyThirtover, 2022-08-21:
Thanks Robert.

  Are you suggesting that I should do something like this:

'''
from machine import WDT

wdt = WDT(timeout=16777216)

async def monitor():
	while True:
		wdt.feed()
		await asyncio.sleep_ms(8000) #  i.e. 8 seconds, which is shorter than 8.3
		print ('Watchdog period COMPLETE')
'''

Regards

Andy

robert-hh, 2022-08-21:
yes, only that the largest value you can supply to WDT() is 8388 or (2**23 - 1)//1000 The time unit is still ms, but the internal counter runs a 1MHz.
So you have to feed the dog latest after 8 seconds.

dpgeorge, 2022-08-21:
> So the documentation should be clarified.

Yes.  And also probably raise an exception in the WDT constructor if the timeout argument is out of range.

robert-hh, 2022-08-22:
Sure, will do.

dpgeorge, 2022-08-24:
Addressed by 47c45d0e7fda10f4a62b7017d48ae452203b22f0

closed 2022-08-24 by dpgeorge