micropython/micropython #9079: ESP32 can't set AP ssid

liangfyx, 2022-08-21:
* Please search existing issues before raising a new issue. For questions about MicroPython or for help using MicroPython, or any sort of "how do I?" requests, please see the MicroPython Forum (https://forum.micropython.org/) or raise a documentation request instead.

* In your issue, please include a clear and concise description of what the bug is, the expected output, and how to replicate it.

* If this issue involves external hardware, please include links to relevant datasheets and schematics.

* If you are seeing code being executed incorrectly, please provide a minimal example and expected output (e.g. comparison to CPython).

* For build issues, please include full details of your environment, compiler versions, command lines, and build output.

* Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name
 - git commit hash and port/board
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.
Board: ESP32
Firmware: esp32-20220618-v1.19.1.bin
Issue: I refer to the official documentation to enable AP mode, but I cannot set ssid.
Detail:
[image]
[image]

marcidy, 2022-08-21:
use `essid` instead of `ssid`

```
ap.config(essid='ESP-AP')
```

this is a real issue as the docs are out of line with the code, might just be a doc issue though

liangfyx, 2022-08-21:
Thank you, I tested it and it works fine~

closed 2022-08-21 by liangfyx