micropython/micropython #9081: No bin file in build-GENERIC, esptool.py runs on just 'make' command?

Jibun-no-Kage, 2022-08-22:
No bin file in build-GENERIC, esptool.py runs on just 'make' command?  It is my understanding from reading the README.md that 'make' will generate the bin file in build-GENERIC (by default), with no attempt to deploy. However, I am seeing the following behavior:

$ pwd
/home/crosstoolng/micropython/ports/esp8266
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Create build-GENERIC/firmware-combined.bin
  File "/home/crosstoolng/esp-open-sdk/xtensa-lx106-elf/bin/esptool.py", line 128
    print 'Connecting...'
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print('Connecting...')?
make: *** [Makefile:216: build-GENERIC/firmware-combined.bin] Error 1

Why is esptool.py even called?

jimmo, 2022-08-22:
> Why is esptool.py even called?

It's used to create the output bin file via the `elf2image` command. See the relevant snippet of the `Makefile`

```
$(FWBIN): $(BUILD)/firmware.elf
	$(ECHO) "Create $@"
	$(Q)esptool.py elf2image $^
	$(Q)$(PYTHON) makeimg.py $(BUILD)/firmware.elf-0x00000.bin $(BUILD)/firmware.elf-0x[0-5][1-f]000.bin $@
```

The issue is that you're using the (very old) esptool.py from `/home/crosstoolng/esp-open-sdk/xtensa-lx106-elf/bin/`

I'm not sure which instructions you're following but you need to ensure that your (up to date) system esptool.py is used instead. See [this line](https://github.com/micropython/micropython/pull/9080/files#diff-13186cf94a9c9315d68c661931e58a36c5585935304b33fc7f19685ca2ca2cf9R87) of the docs update I submitted this morning that addresses this in those instructions.

The reason that it's failing is because your system Python is python3 and this old version of esptool.py is Python 2. (It's not actually connecting, that's just happens to be the line that the parser fails on).

closed 2022-08-22 by jimmo

Jibun-no-Kage, 2022-08-22:
Yup, I realized how old esptool.py was.  I am following the README.md for MP compiling on GitHub.
Just to test, I installed python2, but I also changed the system alias as well as tried $ make PYTHON=python3, but neither worked of course given the old ESP SDK.

This one fails completely when I tested it a day or so ago...
https://github.com/pfalcon/esp-open-sdk.git

This one works to a point,...
https://github.com/ChrisMacGregor/esp-open-sdk.git

The interesting thing is, the above on also exists as a docker image that is the most recent docker image I have found.  But I have yet to test the docker image.

And as I noted in the forum... these are old but there are even older references out there.  It is a real confusing mess via Google.

Since you have or already have updated documentation in the works, that should help the situation.  :)