micropython/micropython #9087: ARDUINO_PORTENTA_H7: sdram as heap

dmazzella, 2022-08-23:
Hi @iabdalkader, @dpgeorge 
I'm trying to use sdram mounted on the Portenta H7 as board heap.

i saw in `ports/stm32/main.c`:
```
// GC init
gc_init(MICROPY_HEAP_START, MICROPY_HEAP_END);
```

and what i tried to do was just assign into `ports/stm32/boards/ARDUINO_PORTENTA_H7/mpconfigboard.h`:
```
#define MICROPY_HEAP_START                  sdram_start()
#define MICROPY_HEAP_END                    sdram_end()
```

but without success.

Did I do something wrong?
do any other preliminary operations need to be done?

Thank you and greetings,
D.

dmazzella, 2022-08-23:
sorry, solved 

into file: ports/stm32/sdram.c:247
```
#if defined(STM32F7)
/* Enable MPU for the SDRAM Memory Region to allow non-aligned
   accesses (hard-fault otherwise)
   Initially disable all access for the entire SDRAM memory space,
   then enable access/caching for the size used
*/
```

added `#if defined(STM32F7) || defined(STM32H7)` and now it work correctly.

```
MicroPython v1.19.1-315-g13dceaa4e-dirty on 2022-08-23; PORTENTA with STM32H747
Type "help()" for more information.
>>> import machine
>>> machine.info()
ID=21003a00:0d513031:31343531
DEVID=0x0450
REVID=0x2003
S=400000000
H=200000000
P1=100000000
P2=100000000
_etext=813e270
_sidata=813e278
_sdata=24000000
_edata=24000160
_sbss=24004000
_ebss=24018350
_sstack=2407bff8
_estack=2407fff8
_ram_start=24000000
_heap_start=24018350
_heap_end=2407bff8
_ram_end=24080000
qstr:
  n_pool=1
  n_qstr=3
  n_str_data_bytes=19
  n_total_bytes=195
GC:
  8196480 total
  2032 : 8194448
  1=38 2=5 m=42
LFS free: 16728064 bytes
>>>
```

i have created a PR for this #9088

dpgeorge, 2022-08-25:
Fixed by 1fbf0efaebd597207f32a910d635a94ec9f6ef69

closed 2022-08-25 by dpgeorge