micropython/micropython #9099: run esp32spiram-20220117-v1.18.bin firmware  occurred Error 'ESP_ERR_NOT_FOUND'

iBigDane, 2022-08-25:
I use ESP32-WROVER-E mode，firmware is esp32spiram-20220117-v1.18.bin，Running the following program:
`
"""
测试SPI,I2S,UART,Timer所有外设一起使用时，是否正常
"""
from machine import I2S, Pin, SPI
from machine import UART, Timer
import time

use_timer = True

def _cb(value):
  print(value)

print("Wait ...")
spi = SPI(2, baudrate=1000000, polarity=0, phase=0, bits=8, firstbit=SPI.MSB, sck=Pin(14), mosi=Pin(27), miso=Pin(15))
audio_out0 = I2S(0, sck=Pin(4), ws=Pin(5), sd=Pin(19), mode=I2S.RX, bits=32, format=I2S.STEREO, rate=16000, ibuf=40000)
audio_out1 = I2S(1, sck=Pin(2), ws=Pin(13), sd=Pin(18), mode=I2S.TX, bits=32, format=I2S.STEREO, rate=16000, ibuf=40000)
u1 = UART(2, baudrate=115200, tx=21, rx=22)
u2 = UART(1, baudrate=115200, tx=26, rx=4)
if use_timer:
    t1 = Timer(0)
    t1.init(period=50, mode=Timer.PERIODIC, callback=_cb)
print(u1)
print(u2)
time.sleep_ms(3000)
t1.deinit()
print("done")
`
displays the following error:
`
Wait ...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 21, in <module>
OSError: (-261, 'ESP_ERR_NOT_FOUND')
`
Thanks

robert-hh, 2022-08-25:
Executed on a Wemos ESP WROVER board with 1.19.1 firmware. Works fine.

MicroPython v1.19.1-315-g8930ecf1a-dirty on 2022-08-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.

iBigDane, 2022-08-26:
I run the 1.19.1 firmware of the program on the ESP32-WROVER-E module is normal,thank you!

closed 2022-08-26 by robert-hh