micropython/micropython #9100: LAN8720 Can't work on esp32 with ETH_CLOCK_GPIO17_OUT

coinight, 2022-08-25:
I read the source code from network_lan.c from ports/esp32
in line 143
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = self->phy_addr;
    phy_config.reset_gpio_num = self->phy_power_pin;
    self->phy = NULL;
it use default config but many boards' lan8720 is setted with other clock
    like my board is GPIO17
In Arduino the code is write like this(https://github.com/espressif/arduino-esp32/blob/ba6e82c30d31a2ddc1276a045e0c42a29b38869f/libraries/Ethernet/src/ETH.cpp Line:257)
        eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
        mac_config.clock_config.rmii.clock_mode = (eth_clock_mode) ? EMAC_CLK_OUT : EMAC_CLK_EXT_IN;
        mac_config.clock_config.rmii.clock_gpio = (1 == eth_clock_mode) ? EMAC_APPL_CLK_OUT_GPIO : (2 == eth_clock_mode) ? EMAC_CLK_OUT_GPIO : (3 == eth_clock_mode) ? EMAC_CLK_OUT_180_GPIO : EMAC_CLK_IN_GPIO;
        mac_config.smi_mdc_gpio_num = mdc;
        mac_config.smi_mdio_gpio_num = mdio;
        mac_config.sw_reset_timeout_ms = 1000;
        eth_mac = esp_eth_mac_new_esp32(&mac_config);
It enabled clock config with {gpio0:EMAC_TX_CLK,gpio0:CLK_OUT1,gpio16:EMAC_CLK_OUT,gpio17:EMAC_CLK_180}
without clock set lan8720 return OSError: esp_eth_driver_install failed,which can't work
Could micropython support clock config ? It really help a lot.Thanks

Sleepwa1ker, 2022-08-25:
same bug

robert-hh, 2022-08-25:
The ESP32 and LAN8720 transceiver use a common 50MHz clock. Usually that is created by the ESP.  But the  LAN8720 breakouts I used had a clock generator themselves. In that case either the ESP32 must be configured to use the clock from the LAN8720, or the LAN8720 clock must be disabled. The latter is not easy. The link below shows what another person did.
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/