micropython/micropython #9105: Pico W: Joining multicast group doesn't receive messages.

jimmo, 2022-08-25:
Originally reported in the forum: https://forum.micropython.org/viewtopic.php?f=21&t=12965

On the Pico W:

```py
import struct, socket

def inet_aton(addr):
    return bytes(map(int, addr.split(".")))

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.bind(("", 5007))
s.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, struct.pack(">4sI", inet_aton("224.1.1.1"), 0))
print(s.recvfrom(1024))
```

From a PC:

```
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.setsockopt(socket.IPPROTO_IP, socket.IP_MULTICAST_TTL, 2)
s.sendto(b'some data', ('224.1.1.1', 5007))
s.close()
```

This same code works fine when the receiving device is an ESP32. On the Pico nothing is received.

felixdoerre, 2022-08-25:
I can give an "explanation" for that, as I was just working on a similar issue: right now the cyw43-driver used for the network module only listens on its own mac address, the broadcast mac address, and one hardcoded one for mdns: https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c#L1888

For getting traffic directed to `224.1.1.1`, you would need to listen to mac address `01:00:5e:01:01:01` which right now is not possible.

I've just added a PR to cyw43 (https://github.com/georgerobotics/cyw43-driver/pull/24) to implement this mac filtering, together with ipv6 support. 

For the v4 multicast, you'd probably also have to implement an `igmp_mac_filter` https://github.com/lwip-tcpip/lwip/blob/853258fff05badfe55ffd8f39a2a31b920310488/src/include/lwip/netif.h#L381, I'd imagine similar to the one I've implemented for ipv6: https://github.com/georgerobotics/cyw43-driver/pull/24/files#diff-64f03fc421199e03e0db4be6979491a5386b3463410d8e4eeb19517be3c5d4b9R131.

jimmo, 2022-08-26:
Thanks @felixdoerre !

I've done as you suggested and implemented the v4 side in https://github.com/georgerobotics/cyw43-driver/pull/25
This fixes the original report from the forum. Thanks for the v6 PR!

closed 2022-12-09 by dpgeorge