micropython/micropython #9114: Viper bug: integers > 0x3fffffff appear as python objects

rkompass, 2022-08-26:
The bug presumably is a result of the same logic as the now closed #8956 (perhaps the fix is also as simple).

Integers outside the mpy small int range ( > 0x3fffffff) are handled as python objects in viper.
This rather should happen with integers > 0xffffffff.

Code to illustrate the issue:

```
# @micropython.viper
# def largeint(a) -> int:
#     b = a & 0xfffffffe
#     return b                   #  ->  ViperTypeError: return expected 'int' but got 'object'

# print(largeint(3))

@micropython.viper
def largeint2(a):
    b = a & 0xfffffffe           #      works, but b is now a python object rather than a viper integer
    return b

print(largeint2(3))

gv = b'\x03\x00\x00\x00'

@micropython.viper
def and32(p:ptr32):
    p[0] = p[0] & 0xfffffffe              # -> ViperTypeError: can't do binary op between 'int' and 'object'
                                          # but  works with 0x3ffffffe and with int(0xfffffffe)

print('Before: ', int.from_bytes(gv, 'little'))   #  -> 3
and32(gv) 
print('After: ', int.from_bytes(gv, 'little'))    # should be 2, but does not work
```

darconeous, 2022-09-15:
Does MicroPython use tagged pointers? If so, that might be the reason for the limitation.

rkompass, 2023-04-15:
Related: [https://github.com/orgs/micropython/discussions/11259](https://github.com/orgs/micropython/discussions/11259)

peterhinch, 2023-04-16:
This bug can lead to confusing behaviour. The line
```py
d :uint32 = 0xFFFFFFFF
```
silently produces a long int, causing later failure. It would be good if the line either threw an exception or produced a `uint32`.