micropython/micropython #9117: Calling i2c.scan() causes restart on Wemos D1 mini pro

jose1711, 2022-08-26:
The following code copied to REPL
```
from machine import Pin, I2C
i2c = I2C(sda=Pin(4), scl=Pin(5))
i2c.scan()
```

..results in device restart:
```
>>> i2c.scan()

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30716, room 16 
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8000, len 996, room 12 
tail 8
chksum 0x06
load 0x3ffe83f0, len 1080, room 0 
tail 8
chksum 0x11
csum 0x11
���2�N�{��N|�
             l
              d$l`B���
                      ��r�l�N�
                              �n�
                                 d ��{�p��l
                                           l`�n�l�l�

                                                    l`��s�l
                                                           ���
                                                              d$ rl��rl��"�|��B$b��B|����$$b��o�no��lo���$�ll����
                                                                                                                 d �o����cc
                                                                                                                           ld䌎�B���c
      l�bsd{$r�o����|���c�܄$�r����
MicroPython v1.19.1-337-gbd4e45fd6 on 2022-08-26; ESP module with ESP8266
Type "help()" for more information.
```

Issue observed with the latest stable ([v1.19.1 (2022-06-18) .bin](http://micropython.org/resources/firmware/esp8266-20220618-v1.19.1.bin)) and also nightly build ([v1.19.1-337-gbd4e45fd6 (2022-08-26) .bin](http://micropython.org/resources/firmware/esp8266-20220826-unstable-v1.19.1-337-gbd4e45fd6.bin)).

The problem goes away when board is flashed with v.1.18 or when using Arduino IDE to scan I2C bus. Here's an example after flashing with 1.18:
```
MicroPython v1.18 on 2022-01-17; ESP module with ESP8266
Type "help()" for more information.
>>> 
>>> from machine import Pin, I2C
>>> i2c = I2C(sda=Pin(4), scl=Pin(5))
>>> i2c.scan()
[]
```

HW info:
  * Wemos D1 mini pro (clone), all pins unconnected
  * connected via microUSB to a PC

Command used for flashing:
```
esptool.py --port /dev/ttyUSB0 --baud 1000000 write_flash --flash_size=4MB -fm dio 0 FILENAME.bin
```

robert-hh, 2022-08-27:
I can replicate that. The situation changes if I connect the mandatory pull-up resistors to the sda and scl pins. Most likely it's a watchdog reset, caused by a false detection of inifinite clock stretching due to missing pull-up resistors.

jose1711, 2022-08-27:
@robert-hh Thank you for your analysis. So basically what you're saying is that pre-1.19.1 version's behaviour was actually incorrect? If so how come that the watchdog does not kick in when using https://learn.adafruit.com/scanning-i2c-addresses/arduino?

robert-hh, 2022-08-27:
I do not know what the pre-1.19 versions caused to behave different, or why Arduino code works different. Maybe it does not have  a watchdog or not clock stretching support. In any case, I2C requires pull-up resistors for operation. For actual set-up without these you can as well reduce the clock stretching timeout, e.g. with:

`i2c = I2C(sda=Pin(4), scl=Pin(5), timeout=1000)`

That will set the timeout to 1 ms, and then i2c.scan() returns an empty list. The default for timeout is 50000 (50 ms), which is too long. A value of 10000 is still fine. That avoids the reset, but without the pull-up resistors the I2C bus is still non-functional.
For convenience of the users, many breakout boards have these implemented.  The standard 1 transistor level converter also come with pull-up resistors. So it may be that you never noticed them.

jose1711, 2022-08-27:
Thank you @robert-hh. I understand that without those pull-up resistors I2C is de facto broken but at the same time I don't think a board reset is a desired outcome. Using a shorter timeout does fix the issue so maybe it's related to this commit https://github.com/micropython/micropython/commit/7d71ae25edc4b405dd03222f044bc7cf3b77f956 (if that is the case than perhaps ESP8266 port should use a shorter than 50ms timeout by default).

robert-hh, 2022-08-27:
The ESP8266 port uses SoftI2C, and that is common to all ports. At the moment, there is no port-specific timeout foreseen, and 50ms is fine as long as you just go for one address. It's just the scan, that checks 104 addresses in a row, before returning to the main processing loop and feeding the watchdog there, that takes longer than the ESP8266 watchdog period. I might be that resetting is a little bit harsh as reaction, but just for this case it seems not worth to change the mechanism. But the behavior might need a few words of explanation in the documentation - hoping that people read the documents. Missing pull-up circuitry is a common cause of irritation.

jose1711, 2022-08-27:
> But the behavior might need a few words of explanation in the documentation - hoping that people read the documents.

FWIW, I do :smile: 

> Missing pull-up circuitry is a common cause of irritation.

Right, but this could also be a loose connection between a shield (already equipped with pull-ups) and a breadboard for instance.