micropython/micropython #9132: tool/mpremote: Hope improve path format support for Windows.

Wind-stormger, 2022-08-29:
In Windows, whether it is copying a file in an absolute path or a file in a folder in a relative path, an error will occur.

### Discussed in https://github.com/orgs/micropython/discussions/9096

<div type='discussions-op-text'>

<sup>Originally posted by **Wind-stormger** August 25, 2022</sup>

Using mpremote 0.3.0 that has been released in pypi, you will get a similar error when you do the following on Windows PowerShell:

```
PS D:\temp> mpremote connect COM1 cp D:\temp\main.py :        
cp D:\temp\main.py :D:\temp\main.py
PS D:\temp> mpremote connect COM1 ls
ls :
          66 D: emp\main.py
         139 boot.py
PS D:\temp> mpremote connect COM1 cat D: emp\main.py
cat :D:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
The filename on the device is renamed to its full Windows absolute pathname and cannot be read with the "ls" command normally.

In case you get the same error, follow the steps below to remove the incorrect file.
CODE: [SELECT ALL](https://forum.micropython.org/viewtopic.php?f=15&t=12964&p=70721#)

PS D:\temp> mpremote connect COM1 cat D:\temp\main.py
cat :D:\temp\main.py
print ("start")
for i in range(2):
    print(i)
print ("end")
PS D:\temp> mpremote connect COM1 rm D:\temp\main.py 
rm :D:\temp\main.py
PS D:\temp> mpremote connect COM1 ls
ls :
         139 boot.py
```
Copy file from device to local absolute paths , but omit the target filename :
```
PS D:\temp> mpremote connect COM1 cp :main.py D:\temp\        
cp :main.py D:\temp\
Traceback (most recent call last):
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\Scripts\mpremote.exe\__main__.py", line 7, in <module>
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\site-packages\mpremote\main.py", line 569, in main
    do_filesystem(pyb, args)
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\site-packages\mpremote\main.py", line 334, in do_filesystem 
    pyboard.filesystem_command(pyb, args, progress_callback=show_progress_bar)
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\site-packages\mpremote\pyboard.py", line 594, in filesystem_command
    op(src, dest2, progress_callback=progress_callback)        
  File "C:\Users\Wind\AppData\Local\Programs\Python\Python310\lib\site-packages\mpremote\pyboard.py", line 499, in fs_get     
    with open(dest, "wb") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'D:\\temp\\'
```

I think this kind of error is related to the drive letter `D:` and `\` of Windows. 

At present, given a target filename, the above error will not occur.

Currently, in Windows, it is not possible to copy multiple files from an absolute path at the same time without corrupting the filenames, because copying multiple files at the same time cannot specify the target filename one by one.
```
PS D:\temp> mpremote connect COM44 cp D:\temp\main.py D:\temp\main1.py :
cp D:\temp\main.py :D:\temp\main.py
cp D:\temp\main1.py :D:\temp\main1.py
PS D:\temp> mpremote connect COM44 ls
ls :
          66 D: emp\main.py
          66 D: emp\main1.py
         139 boot.py
```

</div>

Wind-stormger, 2022-08-29:
Regarding the path format in Windows, I think this method can be used to judge the current system, and then apply the corresponding path format to operate the file.

```py
import platform
print(platform.system())
 
if(platform.system()=='Windows'):
    pass
elif(platform.system()=='Linux'):
    pass
else:
    pass

```

jimmo, 2022-08-29:
Right. The problem here is that `mpremote` just assumes that the path separator is `"/"`. It should be using `os.path.sep` -- no need to query the OS via `platform`.

The main culprit is here:
https://github.com/micropython/micropython/blob/master/tools/pyboard.py#L579

It needs to use `os.path.sep` when splitting a local src, and then replace all `os.path.sep` with `"/"` when generating the remote name from a local name.

dpgeorge, 2022-09-13:
Should be fixed by 57fd66b80f8352e4859e6b71536b6083f9d7279c

closed 2022-09-13 by dpgeorge