micropython/micropython #9136: Turning off WIFI radio on the Pico W

UsuallyErrored, 2022-08-29:
### Discussed in https://github.com/orgs/micropython/discussions/9135

<div type='discussions-op-text'>

<sup>Originally posted by **UsuallyErrored** August 29, 2022</sup>
I'm trying to use an HTML server, as a temporary initial configuration UI, and then once the post request from the client is received, shut down and deep sleep. I'm chasing efficiency as this is a battery application and Im noticing that  

```
wlan.disconnect()
wlan.active(False)
```
Does not shut down the radio, as I'm .07mA higher than without ever starting it during deep sleep. Doing some searching I've found that this is somewhat to be expected as the esp8266 required additional commands ( https://forum.micropython.org/viewtopic.php?t=2734 ) to be sent to completely shut off the WIFI radio. Basically I'm on the hunt for how to turn this radio off, without completely shutting down the cpu. 

Greatly appreciate any time or info you might give. 
</div>

The recently added 
```
machine.lightsleep()
```
Doesn't seem to shutdown the radio either.

lowfatcode, 2022-08-31:
Similar issue here, any attempt to shutdown the WiFi radio fails and seems to leave it in a state that needs a power cycle to recover otherwise the next time `wlan.active()` is called it locks up completely.

mrlunk, 2023-02-14:
use: wlan.deinit()

The deinit method seems to reset the wifi chip and power it down. 

See here: https://github.com/micropython/micropython/blob/31d7ab327b0da4fe7747aba5590a542b88caa123/drivers/cyw43/cyw43_ctrl.c#L127