micropython/micropython #9159: Add color format supporting bi-color 8x8 matrix display

jose1711, 2022-08-30:
https://learn.adafruit.com/adafruit-led-backpack/bi-color-8x8-matrix?view=all#bi-color-8x8-matrix is an 8x8 matrix display able to render 3 colors as: color1, color2, color1+color2. Internally it is defined as a 8x16 matrix where the first half defines points for the first color and the second half does the same for secondary color.

Currently MicroPython lacks support for this color format hence only a single color can be used. A library for CircuitPython exists: https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/blob/083270d530abb88343ebd674931d25fb2efbca1b/adafruit_ht16k33/matrix.py#L224

jimmo, 2022-08-30:
I think this is more likely something that you could do in Python over two 8x8 GS8 framebuffer. i.e. write a Python class that provided all the same methods as FrameBuffer, but then turned it into two calls. Not sure it's worth adding a whole extra color format to support this?

```py
class Matrix16x8:
  def __init__(self, i2c, addr):
   self.buf = bytearray(16*8 + 1) # cmd + <8 bytes> + <8 bytes>
   self.i2c = i2c
   self.addr = addr

   # Create two framebuffers pointing to the 8-byte regions in self.buf
   mv = memoryview(self.buf)
   self.fb1 = framebuf.FrameBuffer(mv[1:9], 8, 8, framebuf.GS8)
   self.fb2 = framebuf.FrameBuffer(mv[9:17], 8, 8, framebuf.GS8)

   # initialise i2c device (e.g. oscillator, etc)

  def show(self):
    self.buf[0] = 0 # set cmd
    self.i2c.writeto(addr, self.buf)  # write cmd+<8 bytes>+<8 bytes>

  def line(self, x1, y1, x2, y2, col):
    self.fb1.line(x1, y1, x2, y2, col & 0xff)
    self.fb2.line(x1, y1, x2, y2, (col >> 8) & 0xff)
  
  ...etc for other methods
```

jimmo, 2022-08-30:
FWIW, I may have the pixel layout wrong... only skimmed the linked driver. It might need to be a single 16x8 framebuf, and then use e.g. 

```
  def line(self, x1, y1, x2, y2, col):
    self.fb.line(x1, y1, x2, y2, col & 0xff)
    self.fb.line(x1 + 8, y1, x2 + 8, y2, (col >> 8) & 0xff)
```

but you get the idea :)

jimmo, 2022-08-31:
Also just realised these are 1-bit not greyscale, updated the code above.

jose1711, 2022-09-02:
> Not sure it's worth adding a whole extra color format to support this?

Thank you for the example, haven't tested yet but it seems it will work.