micropython/micropython #9176: ESP32 Core panic (LoadProhibited) on soft reboot after SDCard() error

jamesquilty, 2022-09-01:
While testing the fix for #8949 I've observed a reproducible core panic on soft reboot with the latest nightly build using our boards, which use the SPI bus, `slot=2` with non-recommended pins. This is not as serious a problem as the original because the panic causes a reboot. The steps to reproduce and the error output are:

```python
MicroPython v1.19.1 on 2022-08-31; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import machine, os
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
>>> os.mount(sd, '/sd')
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
E (94580) sdspi_host: Bus already initialized. Call `sdspi_host_init_dev` to attach an sdspi device to an initialized bus.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> <ctrl-d>
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401666e1  PS      : 0x00060530  A0      : 0x800d7ce9  A1      : 0x3ffcd120
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbf7a4  A5      : 0x00000008
A6      : 0x00000002  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffcd100
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000027  A13     : 0x00000000
A14     : 0x3ffc3c5c  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x40094ddd  LEND    : 0x40094ded  LCOUNT  : 0xffffffff

Backtrace:0x401666de:0x3ffcd120 0x400d7ce6:0x3ffcd140 0x400db7e5:0x3ffcd160 0x400e1b41:0x3ffcd180 0x400e1c4b:0x3ffcd1a0 0x400e262a:0x3ffcd1d0 0x400d8260:0x3ffcd
250 0x400d82ee:0x3ffcd290 0x400d514d:0x3ffcd2b0

ELF file SHA256: 0a70a5c8f2dd45ab

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
```

RSC-Games, 2022-09-09:
Hello jamesquilty!

It would be great if you could provide a decoded backtrace! I cannot help without one.

jamesquilty, 2022-09-09:
@RSC-Games I'd be happy to... how do I do that?

Searching this repository for Issues containing [decoded backtrace](https://github.com/micropython/micropython/search?q=decoded+backtrace&type=issues) finds four hits (other than this Issue).

<https://github.com/micropython/micropython/issues/5970#issuecomment-619403136> suggests using [ESP32 Exception Stack Backtrace Analyzer](https://github.com/tve/esp32-backtrace) which calls the `xtensa-esp32-elf-gdb` installed by PlatformIO to decode the backtrace. I don't have a firmware build environment installed and configured, so that's a barrier which needs to be overcome. Would there happen to be a Dockerfile or a Docker image with a known-working MicroPython build environment which I could build/pull so I can run `xtensa-esp32-elf-gdb` in a container to decode the backtrace?

<https://github.com/micropython/micropython/issues/4754#issuecomment-490322960> says that if the firmware `.elf` file is available then the backtrace can decoded using the crash dump above. Unfortunately, I used the late-August nightly builds for my tests, only downloading the `.bin` file for flashing to the ESP32, and the `.elf` files for those builds are no longer available for download. This is a lower barrier, I should be able to reproduce the crash with one of the latest nightly builds. If I can provide a crash dump with a link to the `.elf` file corresponding to the software, and assuming that you have the `xtensa-esp32-elf-gdb` available, would this be sufficient for you to decode the backtrace?

RSC-Games, 2022-09-10:
@jamesquilty Unfortunately, right now I do not have access to xtensa-esp32-elf-gdb. I have yet to install it on my windows system. I was using a RPi for a while to compile the code but that went under after a while...

If you install Arduino, install the backtrace tool, use the .elf file you have, assuming the image was compiled with IDF 4.4, I would assume that you could decode most of the IDF-related failures, but the micropython information would remain undecoded.

If there is a Docker image I do not currently know about it, but I'm sure some person created one. It honestly would be nice if micropython had an exception stack decoder site for users who installed the stock firmware and experienced a crash, as these individuals likely do not have a development machine with esp-idf installed.

I have a test device with MPY 1.18 installed, however the PCB hookups for the SD card are standard, except for the CS line (line 2). It deinits properly with soft reboot, and uses slot 3. I will have to investigate this more.

RSC-Games, 2022-09-10:
Based off of your register dump, I can identify a few things:
Exception: LoadProhibited (we're trying to access an invalid memory range.)
PC: 40116de: Valid IRAM/IROM address.
A0 (I believe is the stack pointer): Valid address: (800d7ce9)
EXCVADDR: 0x00000004  <- invalid load address

Something is writing to an IROM address. We're not allowed to store data in IROM.
So what likely happened is the accessed SDCard object is a nullptr but for some reason the soft reboot code thinks the sdcard has been initialized. So the call likely fails in sd.deinit().

RSC-Games, 2022-09-12:
I found a little more information here at https://github.com/micropython/micropython/blob/master/ports/esp32/machine_sdcard.c
The function sd_deinit appears to be the culprit.

```
STATIC mp_obj_t sd_deinit(mp_obj_t self_in) {
    sdcard_card_obj_t *self = self_in;

    DEBUG_printf("De-init host\n");

    if (self->flags & SDCARD_CARD_FLAGS_HOST_INIT_DONE) {
        #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 2, 0)
        if (self->host.flags & SDMMC_HOST_FLAG_DEINIT_ARG) {
            self->host.deinit_p(self->host.slot);
        } else
        #endif
        {
            self->host.deinit();
        }
        if (self->host.flags & SDMMC_HOST_FLAG_SPI) {
            // SD card used a (dedicated) SPI bus, so free that SPI bus.
            spi_bus_free(self->host.slot);
        }
        self->flags &= ~SDCARD_CARD_FLAGS_HOST_INIT_DONE;
    }

    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(sd_deinit_obj, sd_deinit);
```

So, from examining the file, there are a few points where the null read could happen:
```
if (self->host.flags & SDMMC_HOST_FLAG_DEINIT_ARG) {
     self->host.deinit_p(self->host.slot);
} 
```
```
 if (self->host.flags & SDMMC_HOST_FLAG_SPI) {
     // SD card used a (dedicated) SPI bus, so free that SPI bus.
     spi_bus_free(self->host.slot);
}
```

This is based on the fact that the crash really is happening in machine_sdcard.c. It could actually be in the SPI driver.

dpgeorge, 2022-09-13:
I can't reproduce this crash.  From your log it looks like you tested using v1.19.1, not the latest master.

Here's my test, on Lolin 32 Pro:
```
MicroPython v1.19.1-394-g2d4e7e99b on 2022-09-13; ESP32 module with ESP32
Type "help()" for more information.
>>> import machine
>>> sd=machine.SDCard(slot=2, sck=14, miso=2, mosi=15, cs=13)
>>> import os
>>> os.mount(sd, '/sd')
>>> sd=machine.SDCard(slot=2, sck=14, miso=2, mosi=15, cs=13)
E (329530) sdspi_host: Bus already initialized. Call `sdspi_host_init_dev` to attach an sdspi device to an initialized bus.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> 
MPY: soft reboot
MicroPython v1.19.1-394-g2d4e7e99b on 2022-09-13; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```
Also similar thing using default slot=1 SDIO:
```
MicroPython v1.19.1-394-g2d4e7e99b on 2022-09-13; ESP32 module with ESP32
Type "help()" for more information.
>>> import os, machine
>>> sd=machine.SDCard()
>>> os.mount(sd, '/sd')
>>> sd=machine.SDCard()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> 
MPY: soft reboot
MicroPython v1.19.1-394-g2d4e7e99b on 2022-09-13; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```

RSC-Games, 2022-09-13:
Since I apparently disabled machine.SDCard support on my board when I compiled my code, I was unable to reproduce the crash. If it is possible to migrate your code base to the master branch, could we close this?

RSC-Games, 2022-09-13:
That is very helpful. So it does appear to be an issue in IDF. If it would be possible, could you paste only the backtrace in the decoder? This would prevent it from decoding the address in the PC, which while important, is usually very close to that of the top entry on the stack.

Anyway, here's line 574 in spi_common.c:
https://github.com/espressif/esp-idf/blob/master/components/driver/spi_common.c

```
    uint32_t missing_flag = flags & ~temp_flag;
    missing_flag &= ~SPICOMMON_BUSFLAG_MASTER;//don't check this flag

    if (missing_flag != 0) {
    //check pins existence
        if (missing_flag & SPICOMMON_BUSFLAG_SCLK) {
            ESP_LOGE(SPI_TAG, "sclk pin required.");
        }
        if (missing_flag & SPICOMMON_BUSFLAG_MOSI) {
            ESP_LOGE(SPI_TAG, "mosi pin required.");
        }
        if (missing_flag & SPICOMMON_BUSFLAG_MISO) {
            ESP_LOGE(SPI_TAG, "miso pin required.");
        }
        if (missing_flag & SPICOMMON_BUSFLAG_DUAL) {  // <-- Causes crash 
            ESP_LOGE(SPI_TAG, "not both mosi and miso output capable");
        }
        if (missing_flag & SPICOMMON_BUSFLAG_WPHD) {
            ESP_LOGE(SPI_TAG, "both wp and hd required.");
        }
        if (missing_flag & SPICOMMON_BUSFLAG_IOMUX_PINS) {
            ESP_LOGE(SPI_TAG, "not using iomux pins");
        }
```

jamesquilty, 2022-09-13:
I just tried pasting only the backtrace, but for some reason there's no output when its done that way:

```shell
root@50c4d0359ab2:/MicroPython# ./esp32-backtrace esp32spiram-20220913-unstable-v1.19.1-395-g57fd66b80.elf
reading backtrace from stdin
Backtrace:0x40166752:0x3ffcd120 0x400d7cf2:0x3ffcd140 0x400db7f1:0x3ffcd160 0x400e1b4d:0x3ffcd180 0x400e1c57:0x3ffcd1a0 0x400e2636:0x3ffcd1d0 0x400d826c:0x3ffcd250 0x400d82fa:0x3ffcd290 0x400d5159:0x3ffcd2b0
^D
root@50c4d0359ab2:/MicroPython#
```

RSC-Games, 2022-09-13:
If you omit the "Backtrace:" will it work?

jamesquilty, 2022-09-13:
Yes, it does work when "Backtrace:" is omitted! That's a bit weird of `exp32-backtrace`, but thanks to your suggestion the decoded backtrace is:

```shell
BT-0: 0x40166752 is in spi_bus_free (/home/micropython/esp-idf-v4.2/components/driver/spi_common.c:571).
BT-1: 0x400d7cf2 is in sd_deinit (/home/micropython/micropython-autobuild/ports/esp32/machine_sdcard.c:288).
BT-2: 0x400db7f1 is in fun_builtin_1_call (/home/micropython/micropython-autobuild/py/objfun.c:71).
BT-3: 0x400e1b4d is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:693).
BT-4: 0x400e1c57 is in mp_call_function_1 (/home/micropython/micropython-autobuild/py/runtime.c:671).
BT-5: 0x400e2636 is in mp_call_function_1_protected (/home/micropython/micropython-autobuild/py/runtime_utils.c:33).
BT-6: 0x400d826c is in gc_collect_end (/home/micropython/micropython-autobuild/py/gc.c:386).
BT-7: 0x400d82fa is in gc_sweep_all (/home/micropython/micropython-autobuild/py/gc.c:502).
BT-8: 0x400d5159 is in mp_task (/home/micropython/micropython-autobuild/ports/esp32/main.c:209).
```

RSC-Games, 2022-09-15:
I did some more digging. I was annoyed by the fact that I could not find spi_common_internal.h, so I downloaded IDF and looked through the modules. I found the module, then realized that I had used the wrong branch for finding the information. I switched to IDF 4.2 (as your decoded backtrace says IDF 4.2, not 4.4), and the output made a little bit more sense. @jamesquilty If you could post the IDF 4.2 spi_common.c file in the docker image that would be great! I imagine the line number is slightly off from the latest 4.2 commit and the docker image you have.

BT-0 with IDF 4.2 spi_common.c:
```
esp_err_t spi_bus_free(spi_host_device_t host_id)
{
    if (bus_ctx[host_id] == NULL) {
        return ESP_ERR_INVALID_STATE;  // <-- stack entry here.
    }

    esp_err_t err = ESP_OK;  // <-- crash occurs here.
    spicommon_bus_context_t* ctx = bus_ctx[host_id];
    spi_bus_attr_t* bus_attr = &ctx->bus_attr;

    if (ctx->destroy_func) {
        err = ctx->destroy_func(ctx->destroy_arg);
    }

    spicommon_bus_free_io_cfg(&bus_attr->bus_cfg);

#ifdef CONFIG_PM_ENABLE
    esp_pm_lock_delete(bus_attr->pm_lock);
#endif
    spi_bus_deinit_lock(bus_attr->lock);

    free(bus_attr->dmadesc_rx);
    free(bus_attr->dmadesc_tx);

    if (bus_attr->dma_chan > 0) {
        spicommon_dma_chan_free (bus_attr->dma_chan);
    }
    spicommon_periph_free(host_id);

    free(ctx);
    bus_ctx[host_id] = NULL;
    return err;
}
```

jamesquilty, 2022-09-18:
@RSC-Games, I found the file at `/opt/esp/idf/components/driver/spi_common.c` in the [espressif/idf:release-v4.2](https://hub.docker.com/layers/espressif/idf/release-v4.2/images/sha256-f308bc2e67dce0f0816ce025d0ae26568435d3b049840be5b5d5ffe2a93fe3df?context=explore) image. The `spi_bus_free()` function definition is identical, the whole file is attached: [spi_common.c.zip](https://github.com/micropython/micropython/files/9595320/spi_common.c.zip)

jamesquilty, 2022-09-27:
@RSC-Games Just a quick note to say I'm willing (happy, even 😄) to build the firmware with the modification you suggest, but it will take me a while before I'll be able to spend the time to set up a firmware build environment. FWIW, I'll most likely create the firmware build environment as a Docker image, which should have the benefit that you can run the same build as me by using the same image.

RSC-Games, 2022-09-30:
@jamesquilty I was a little busy the past couple of days. I finally have time to install the ESP-IDF toolchain on my development machine, so I should be able to decode backtraces. As for testing the crash, that may be a little difficult because all of my ESP32 devices are on perfboard with the SPI bus hooked up with the standard pins. One of them uses non-standard pins but I "forgot" to compile machine.SDCard support into the binary. I'm not sure how much else I can debug the crash other than testing and re-running the code to see if compiling differently influences the crash location. After that, all we can do is figure out if this is an IDF problem or some obscure bug in MicroPython.

projectgus, 2025-02-11:
Thanks for being patient while someone got around to looking at this again.

I'm unable to reproduce this problem running the provided code on master branch:

```
MicroPython v1.25.0-preview.283.g11c9656fad on 2025-02-11; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> import machine, os
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
>>> os.mount(sd, '/sd')
>>> sd = machine.SDCard(slot=2, sck=13, miso=34, mosi=23, cs=12)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> 
MPY: soft reboot
MicroPython v1.25.0-preview.283.g11c9656fad on 2025-02-11; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> 
```

I'm not sure what the fix was, but it seems to be fixed in any case.