micropython/micropython #9186: popitem() consistently return (key, value) pair in FIFO order instead of LIFO (python >= 3.7) or arbitrary (python<3.7

pidou46, 2022-09-01:
Extract from python doc:
```
popitem()

    Remove and return a (key, value) pair from the dictionary. Pairs are returned in LIFO order.

    [popitem()](https://docs.python.org/3.8/library/stdtypes.html#dict.popitem) is useful to destructively iterate over a dictionary, as often used in set algorithms. If the dictionary is empty, calling [popitem()](https://docs.python.org/3.8/library/stdtypes.html#dict.popitem) raises a [KeyError](https://docs.python.org/3.8/library/exceptions.html#KeyError).

    Changed in version 3.7: LIFO order is now guaranteed. In prior versions, [popitem()](https://docs.python.org/3.8/library/stdtypes.html#dict.popitem) would return an arbitrary key/value pair.
```

```
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> test.popitem()
(0, 2)
>>> test.popitem()
(1, 2.5)
>>> test.popitem()
(2, 2)
>>> test.popitem()
(3, 4)
>>> 

```

```
Python 3.9.9 (/usr/bin/python3)
>>> 
>>> test.popitem()
(3, 4)
>>> test.popitem()
(2, 2)
>>> test.popitem()
(1, 2.5)
>>> test.popitem()
(0, 2)
```

jimmo, 2022-09-01:
> Changed in version 3.7: LIFO order is now guaranteed.

This change in behavior of `popitem` was a consequence of Python 3.7's change to make dictionaries preserve insertion order.

MicroPython has not adopted this change as it makes dictionaries take more RAM. (Although see https://github.com/micropython/micropython/pull/6173 for discussion).

If you need to preserve insertion order, and therefore make popitem behave the way you're expecting, then you can explicitly use `collections.OrderedDict` (which is available on many ports including ESP32).

closed 2022-09-01 by jimmo

pidou46, 2022-09-02:
Thank you, sorry for not having searching enough to found #6173

I have found a workaround