micropython/micropython #9197: Issues Reading Files between 7/20 build & 8/30

michaeka79, 2022-09-02:
I am trying to do a very simple open a file and read it process. I have a private key from Amazon that was originally encoded as a PEM key but I converted it into a DER key.  with v1.19.1-192 built on 7/20/2022, I am able to open this DER file and read the key into a variable. Starting with the build on 8/30/2022 v1.19.1-345 I get a UnicodeError when I try to read the files.  I have a PEM file for Amazon's CA that reads without an issue it is just the DER key files that appear to now be broken. I'll gladly call myself a noob in this area, but it doesn't seem like anything changed in the coding of Micropython since then that would have caused this kind of issue.  But I'm totally stumped!

dlech, 2022-09-02:
If this is a binary file, open it with `'rb'` instead of `'r'`. If this is a text file, it contains invalid unicode characters.

dlech, 2022-09-02:
FWIW, the change that triggered this is #9083 .

michaeka79, 2022-09-02:
I'll gladly accept my noob badge for this one. Thank you! Weird you'd want to be using safe Unicode characters ;)

jimmo, 2022-09-03:
Thanks @dlech !

closed 2022-09-03 by jimmo