micropython/micropython #9204: tools/mpremote Copy folder content creates folder on board

lhoff94, 2022-09-03:
Hi, 

i noticed a behaviour with mpremote that doesn't seem right. I try to copy the content of a complete folder to the board. But instead of putting the folder content in the root of the device it creates a folder on the device and copies everything there. 

## Example:
Assume if have to following folder setup:
```
~/test$ tree mpy-project/
mpy-project/
├── boot.py
├── lib
│   └── urequests.py
└── main.py
```
I want to put the content of mpy-project on my empty board.
My first attempt was using this command:
`mpremote connect /dev/ttyUSB1 cp -r mpy-project/* :`
It transferred all files to the board but it was all in a folder called "mpy-project". It looked like that:
```
~/test$ mpytool -vp /dev/ttyUSB0 tree
./
└─ mpy-project/
├─ lib/
│  └─ urequests.py
├─ boot.py
└─ main.py
```
That seems counterintuitive for me since the `*`should be expanded and then the files should be copied. 
There is also no use case where that behavior would make sense. If I wanted the complete folder copied I would leave `/*`out.

Or do i miss something?

## Information about the version of MicroPython you're running
- Micropython 1.19.1
- mpremote 0.3.0
- ESP32

hubua, 2023-09-09:
I also belevie this is not a right behavior, and should be fixed.
In my case, assuming this folder sctuctire
```
├───Blink
│       boot.py
│       main.py
│       secrets.json
```
When executing the `mpremote cp -r ./Blink/* :` command, I'm getting an error alltogether
`The filename, directory name, or volume label syntax is incorrect: './Blink/*'`

peterhinch, 2023-09-09:
Under Linux, if `foo` is a subdirectory of the current directory
```bash
$ cp -r foo/* ~/temp
```
copies the contents of `foo` to `~/temp` while
```bash
$ cp -r foo ~/temp
```
copies the directory and its contents. I can confirm that `mpremote` copies the directory and contents in both cases. It would be good if it conformed with the Unix syntax.
@hubua 
I can't replicate this.
```bash
$ mpremote cp -r ./Blink/* :
```
copies the directory and contents to the device with no syntax error. Are you running an old version?
```bash
[adminpete@meerkat]: ~
$ mpremote --version
mpremote 1.20.0
```

jimmo, 2023-09-11:
FYI the `mpremote cp -r` behavior is going to get an update shortly: see https://github.com/micropython/micropython/pull/11777

 I spent some time in that PR (and wrote some tests) to ensure that the recursive copy behavior matched the standard behavior of the `cp` tool on Linux. (But can't comment on the existing behavior)

sambeard, 2024-05-31:
> Under Linux, if `foo` is a subdirectory of the current directory
> 
> ```shell
> $ cp -r foo/* ~/temp
> ```
> 
> copies the contents of `foo` to `~/temp` while
> 
> ```shell
> $ cp -r foo ~/temp
> ```
> 
> copies the directory and its contents. I can confirm that `mpremote` copies the directory and contents in both cases. It would be good if it conformed with the Unix syntax. @hubua I can't replicate this.

On my windows machine I get a WinError (WinError123) when attempting to perform a copy operation using the wildcard syntax like so

``` shell
> mpremote cp ./src/* :
cp ./src/* :
mpremote: [WinError 123] The filename, directory name, or volume label syntax is incorrect: './src/*'
```
Assuming that the issue is that windows does not recognize wildcards in the path syntax

sandyscott, 2024-12-09:
The syntax to copy the contents of a folder into the root of a device without creating a folder is:
`mpremote cp -r source/. :`

Commenting here because this was the first result I found when I wanted to the do the same thing. I found the solution in the bug referenced above https://github.com/micropython/micropython/pull/11777#issuecomment-1702075572

As a minimum a documentation tweak would help, though I agree the choice of `.` as the wildcard indicator is more confusing than `*` as suggested by @lhoff94