micropython/micropython #9205: unpacking stared arguments

kr-g, 2022-09-04:
unpacking stared arguments does not comply to cpython
this happens also with normal assignment

micropython esp8266 and esp32

```
>>> a= [1,2,3]
>>> print(a)
[1, 2, 3]
>>> print(*a)
1 2 3
>>> print([*a])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: *x must be assignment target
>>> 
```

cpython

```
>>> a= [1,2,3]
>>> print(a)
[1, 2, 3]
>>> print(*a)
1 2 3
>>> print([*a])
[1, 2, 3]
>>> 
```

jimmo, 2022-09-04:
This feature is known as "Additional Unpacking Generalizations" and was described in PEP 448 (https://peps.python.org/pep-0448/) and added in Python 3.5. MicroPython's baseline is 3.4 (although we do support a number of later features, and this would be a nice one to add especially as it starts becoming more common in everyday Python code).

We're already tracking this on the differences list: https://docs.micropython.org/en/latest/differences/python_35.html

closed 2022-09-04 by jimmo

dlech, 2022-09-04:
There is some relevant discussion at https://github.com/micropython/micropython/issues/1329#issuecomment-1084815423 as well.