micropython/micropython #9208: ports/rp2: Extend "Flash" to map into arbitrary slices of flash memory.

Gadgetoid, 2022-09-04:
Currently RP2's `rp2.Flash()` returns a block device that affects the entire, compile-time determined user flash region of the Pico/RP2040 board.

This- insofar as I'm aware- precludes multiple filesystems from existing alongside each other in flash, since there's no way to tell `os.VfsLfs2.mkfs()` *where specifically* to put its filesystem. I'd like to "fix" this.

Right now `rp2.Flash()` is a singleton class: https://github.com/micropython/micropython/blob/master/ports/rp2/rp2_flash.c

I propose upgrading it to a regular, instanced class with two constructor arguments: `base` (or `offset` probably) and `size`.

The default behavior for `rp2.Flash()` would be to use the entire storage region, matching its current behavior.

The class already has `flash_base` and `flash_size` to store the settings, and observes these values. I think the changes would be pretty minimal.

The endgame would be to make something like the following `_boot.py` work:

```
import os
import machine, rp2

# Try to mount the filesystem, and format the flash if it doesn't exist.
# Note: the flash requires the programming size to be aligned to 256 bytes.

USER_FLASH_SIZE = rp2.Flash().ioctl(4, 0) * rp2.Flash().ioctl(5, 0) # Extremely cursed
CONFIG_FS_SIZE = 48 * 1024

bdev_root = rp2.Flash(0, USER_FLASH_SIZE - CONFIG_FS_SIZE)
try:
    vfs_root = os.VfsLfs2(bdev_root, progsize=256)
except:
    os.VfsLfs2.mkfs(bdev_root, progsize=256)
    vfs_root = os.VfsLfs2(bdev_root, progsize=256)
os.mount(vfs_root, "/")

bdev_config = rp2.Flash(USER_FLASH_SIZE - CONFIG_FS_SIZE, CONFIG_FS_SIZE) 
try:
    vfs_config = os.VfsLfs2(bdev_config, progsize=256)
except:
    os.VfsLfs2.mkfs(bdev_config, progsize=256)
    vfs_config = os.VfsLfs2(bdev_config, progsize=256)
os.mount(vfs_config, "/config")

del os, bdev_root, bdev_config, vfs_root, vfs_config, CONFIG_FS_SIZE
```

The above example shows the use of an additional "config" filesystem right at the end of user flash. The intention here is to provide a region in flash that firmware upgrade .uf2 files with pre-baked filesystems do not touch so that user config can be preserved.

See: https://github.com/pimoroni/enviro/releases/tag/v0.0.7 for an example of a firmware that includes a fully initialized and populated LittleFS filesystem in the .uf2 file.

Other uses for this might be a split "user" and "logging" filesystem using FatFS for mass storage mountable user files, and LittleFS for highly robust and fault tolerant logging.

I intend to try and make and test the above changes, but I'm raising this as an issue to invite discussion about:
1. Whether this is a good idea
2. If there's a better way (some secret VFS sauce I don't understand)
3. If removing the singleton approach in `rp2.Flash()` would be dangerous/breaking somehow
4. Other!

Gadgetoid, 2022-09-04:
This proposal would conflict with the changes proposed in https://github.com/micropython/micropython/pull/8381

dpgeorge, 2022-09-05:
The stm32 port already supports this feature.  The default `pyb.Flash()` returns access to the entire available flash, but you can use `pyb.Flash(start=..., len=...)` to get a slice of the flash (`start` and `len` are measured in bytes).  See docs here: http://docs.micropython.org/en/latest/reference/filesystem.html#hybrid-stm32

> Whether this is a good idea

Yes I think it's a good idea!  And I think it's best to copy the stm32 API, ie use `start` and `len`.

> If there's a better way

There is an alternative: you can create a custom Python block device class that implements `readblocks`, `writeblocks` and `ioctl`, and uses `rp2.Flash()` as the underlying storage but partitions it up.  This won't be as efficient as `rp2.Flash(start, len)` implement in C.

> If removing the singleton approach in `rp2.Flash()` would be dangerous/breaking somehow

It would be good to retain a static instance for the entire flash, but create new instances on the heap if start/len is not the entire flash.

> is there some scope here to define partitions at board level

We may eventually need a general concept of "partition" which is the same across ports.  Eg reserve the first page of flash to store the partition table.  This would mean firmware can easily auto-detect the filesystem layout even if everything (except the partition table) is erased.  But that approach would still need a flexible `rp2.Flash(start, len)` to actually mount/mkfs the partitions.

Gadgetoid, 2022-09-05:
Oh wow, the esp32 also has a concept of partitions and I should definitely look across ports to see if there's already an established pattern :laughing: sorry, I get struck with RP2 tunnel vision far too often.

It might even be possible to steal some flash from firmware, to retrospectively add a partition table to Pico ports without it being a breaking, filesystem-corrupting change. But I'm getting ahead of myself here!

Tweaking `rp2.Flash()` as you suggest so that we can just roll our own `_boot.py` with multiple partitions would be a good first step, I think?

I've raised a PR to this end, so we have some (working, insofar as I can tell from my tests!) code to talk around - https://github.com/micropython/micropython/pull/9213

dpgeorge, 2022-09-08:
Constructing flash slices was added in 6aa3c946347281875165392c09753547d8c77fc3

@Gadgetoid can this issue be closed?

Gadgetoid, 2022-09-12:
Yes.

There's probably a discussion to be had around board-specific partition setups, but that's for another time and place.

closed 2022-09-12 by dpgeorge