micropython/micropython #9209: ESP32S3 Pin 44 IRQ not functioning properly

msamblanet, 2022-09-04:
On an ESP32S3 (Unexpected Maker ProS3) board, I have a pushbutton connected to GPIO 44 (input, pullup) and configure an IRQ_FALLING interrupt to it.   The expectation is that my IRQ handler will be called for every falling edge.

- In v1.19.1 release, this works as expected, calling the IRQ handler for every falling edge
  - Downloaded from: https://micropython.org/resources/firmware/pros3-20220618-v1.19.1.bin
  - MicroPython v1.19.1 on 2022-06-18; ProS3 with ESP32-S3

- In v1.19.1 unstable (2022-08-29), the IRQ handler is called once and then never again
  - Downloaded from https://micropython.org/resources/firmware/pros3-20220829-unstable-v1.19.1-339-g9a826e0f2.bin
  - MicroPython v1.19.1 on 2022-08-29; ProS3 with ESP32-S3

Sample code that demonstrates this issue:

```
from machine import Pin
from time import sleep_ms

btn = Pin(44, Pin.IN, Pin.PULL_UP)

btn_cnt = 0
def irqHandler(_):
  global btn_cnt
  btn_cnt += 1

btn.irq(trigger = Pin.IRQ_FALLING, handler = irqHandler)

while True:
  print(btn_cnt)
  # On 1.19.1 release, this increases on each click
  # On 1.19.1-20220829, this increments to 1 and then stops
  
  sleep_ms(250)
```

UnexpectedMaker, 2022-09-30:
IO44 is the UART0 RX pin, so maybe something changed with UART0 config since 1.19.1 - Have a look through the commits and you might find something that's effecting the use of that IO.

dpgeorge, 2022-09-30:
As far as I can tell, the REPL is not enabled on this board on the UART.

It may be that the IDF version changed in the two builds referenced above.  @msamblanet can you run:
```py
import platform
print(platform.platform())
```
and share the result for both the working and non-working firmware?

msamblanet, 2022-10-01:
@dpgeorge  - Sorry for the delay - I just got back from 3 weeks of travel.  

I would need to check notes but I believe I tested both downloaded and self-built versions of the firmware with the same results, and I know the binaries I built were the same IDF.

I will re-test after I get back in order from my travels (likely 3-5 days) and report back along with the platform and IDF info.

As a side note, I have worked around this with a bodge wire to another GPIO but for layout reasons, IO44 was much more convenient so it would be nice to get it working.  I am not using UART0 for anything in my project.

msamblanet, 2022-10-18:
@dpgeorge Sorry for the delay again - I came down with covid...finally recovered and caught up from backlog...

I just did fresh tests...

- Test process:
  - Download test firmware from: https://micropython.org/download/pros3/
  - reset into flash mode
  - esptool erase_flash
  - esptool write_flash -z 0x0 pros3-xxxxxx.bin
  - reset into running mode
  - mpremote cp main.py :
    - main.py is the exact code from my original post
  - mpremote
  - ctrl-d to restart
  - click right button multiple
  - ctrl-c to break
  - import platform
  - print(platform.platform())
- Results:
  - Nightly build:
    - pros3-20221014-unstable-v1.19.1-544-g89b320737.bin
    - Fails as described above (0s until first click then 1 but never advances past 1)
    - MicroPython-1.19.1-xtensa-IDFv4.4.2-with-newlib3.3.0
  - Stable build:
    - pros3-20220618-v1.19.1.bin
    - Works as expected (value increments every click by random amount as there is no debounce)
    - MicroPython-1.19.1-xtensa-IDFv4.4.1-with-newlib3.3.0