micropython/micropython #9226: Wemos D1 Mini (esp8266) continuous boot loop due to WDT Reset

gururise, 2022-09-06:
After flashing with:
> esptool.py --port /dev/ttyUSB0 --baud 1000000 write_flash --flash_size=4MB --flash_mode dio 0 esp8266-20191220-v1.12.bin

All versions of Micropython greater than 1.13 will continually boot loop due to a WDT timer reset on a Wemos D1 Mini (4MB Flash):
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30712, room 16 
tail 8
chksum 0x16
load 0x3ffe8000, len 996, room 0 
tail 4
chksum 0x70
load 0x3ffe83f0, len 1064, room 4 
tail 4
chksum 0xa3
csum 0xa3
o{g|d$l$`;$ll`sldd`{$l$`{$;$cbbbrsc'o'ld$d$dncddspc$c{${l{on#c8
MicroPython v1.15 on 2021-04-18; ESP module with ESP8266
Type "help()" for more information.
```
This boot loop makes it impossible to interact with the REPL.
On v 1.13, it still boot loops; however, it takes about 10 seconds before the WDT resets the chip, and in those 10 seconds, you can interact with the REPL, and as long as you are interacting the WDT will not reset.

MeckPommER, 2022-09-06:
Just tried with Thonny to get my Wemos D1 mini up and running with the current version 1.19.1 - no problem.

gururise, 2022-09-09:
Strange.  I tried reducing baud to 112500 and also tried Thonny with the same result. I also tried a few other MCU's:

1. Wemos D1 Mini - 4MB v1.19.1 - boot loop due to wdt reset
2. ESP-01S - 1MB v1.19.1 - boot loop due to wdt reset
3. ESP-01 - 512KB v1.19.1 - boot loop due to wdt reset
4. Hitec ESP32 w/OLED - v1.19.1 - worked perfectly

I used the same microUSB cable for all three MCU's, so I dont believe its the cable.  Something strange is going on with the ESP8266...

robert-hh, 2022-09-09:
Maybe Thonny uses a different flash mode for flashing. See the -fm or --flash_mode options.

scruss, 2022-10-10:
@gururise - please see this discussion item — [After flashing cannot enter to programming even Repl does not answer](https://github.com/micropython/micropython/discussions/9523#discussioncomment-3830161) — where I outline working options for the LOLIN/Wemos D1 Mini v4. The esptool command lines I used to erase flash and program the binary are quite different from yours.

robert-hh, 2022-10-10:
ESP01 1M and ESP01 512k require to use the specific 1M and 512k firmware images.

gschwaer, 2023-09-13:
I had the same issue and found a solution [here](https://forum.micropython.org/viewtopic.php?t=9046).

**TLDR** It seems to be a hardware bug, where enabling the wifi will trigger a reset due to unstable supply power. There is a software solution and hardware solution:

**Software Solution**

(I tried this and it worked)

In the short time frame after a reset when the REPL works, paste the following lines (including the newline) into the REPL:
```python
import network; network.WLAN(network.AP_IF).active(False)

```
This will deactivate the wifi and the board will not reboot. You can make this permanent by putting this line in your `main.py`. It was reported that STA mode does not have this issue, so if you don't need AP mode, this may already be good enough for you.

**Hardware Solution**

(I did not try that yet, I plan to report back)

Connect a 100uF capacitor between EN and GND. This was reported as not working (at least for some) in the linked post.

gschwaer, 2023-09-25:
Today I flashed micropython on a custom board I made a few years ago which uses a ESP-12F module. Same problem. Same fix works: disabling the AP at boot. :thinking: Interestingly before that, I had NodeMCU running on this board and iirc the AP worked fine.