micropython/micropython #9230: TypeError: expecting an SPI object

Meekdai, 2022-09-07:
I compiled the latest micropython firmware for my custom board with W5500 .
Compilation process:
https://github.com/Meekdai/micropython-actions/actions/runs/3005187684

After download firmware , it can run normally.

```
MicroPython v1.19.1-375-ge90b85cc9 on 2022-09-07; MP405WS with STM32F405RG
Type "help()" for more information.
>>>
```

But an error occurs when using network.WIZNET5K.

```
>>> import network
>>> nic=network.WIZNET5K(pyb.SPI(2), pyb.Pin.board.Y5, pyb.Pin.board.Y1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: expecting an SPI object
>>>
```

robert-hh, 2022-09-07:
You have to use a machine.spi object.

Meekdai, 2022-09-07:
Thanks @robert-hh 
It works with machine.spi.

So the following documentation needs to be modified
https://docs.micropython.org/en/latest/library/network.WIZNET5K.html