micropython/micropython #9232: docs: RP2 DS18X20 examples do not work with latest nightly builds

msetina, 2022-09-07:
Checking latest nightly builds and found missing ds18x20. Going over code I could find:
options.defaults(ds18x20=False)
module("onewire.py", opt=3)
if options.ds18x20:
    module("ds18x20.py", opt=3)

Is this a plan or just current misconfiguration?
Will there be a config tool for firmware?

jimmo, 2022-09-08:
Thanks @msetina for raising this. I made a mistake in a recent refactoring with the rule for whether the "onewire" driver implicitly includes the "ds18x20" driver.

Fixed by #9244 (or #9220 which is the next step of that refactoring)

dpgeorge, 2022-09-08:
> Will there be a config tool for firmware?

In short, no.  But you can easily copy the missing .py files to the device's filesystem to have them available.

msetina, 2022-09-08:
Fixed in rp2-pico-w-20220908-unstable-v1.19.1-378-g24678fe45

closed 2022-09-08 by msetina