micropython/micropython #9233: uasyncio StreamWriter concurrency failure

peterhinch, 2022-09-07:
The following script produces unexpected behaviour on a Pyboard 1.1
```python
import uasyncio as asyncio
from machine import UART
import monitor
from pyb import LED
led = LED(1)

uarts = []  # Pins x9, x3, y9, y1
for n in (1, 2, 3, 6):
    uarts.append((UART(n, 9600, timeout=0), n))

async def sender(u, n):
    s = f"Hello from UART {n} "# * 55  # 990 chars
    swriter = asyncio.StreamWriter(u, {})
    while True:
        swriter.write(s)
        await swriter.drain()
        await asyncio.sleep(0)

async def heartbeat():
    while True:
        await asyncio.sleep_ms(100)
        led.toggle()

async def main():
    for v in uarts:
        asyncio.create_task(sender(*v))
    await heartbeat()

asyncio.run(main())
```
The code behaves as if writes to separate UARTs were controlled by a mutex. There is no concurrency. 
[image]

With 990 character strings (see code comment), the heartbeat LED changes state only once per second, implying that the scheduler is locked for the duration of each write. The LA trace is similar to above, but each burst is ~1s long.

peterhinch, 2022-09-10:
The [docs](http://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.write) for `uart.write` are terse and unclear on the subject of blocking. On STM even with timeouts set to zero it seems that `uart.write(buf)` is designed to block until the entire buffer is written out. A timeout will only occur if this is prevented by hardware flow control. I don't know if this behaviour is consistent across ports.

To be compatible with `StreamWriter` a device should support a `.write(buf)` method which sends as much data as it can without blocking. It returns the number of bytes sent.

If I'm right, there is a dilemma: to change `uart.write` to be compatible with `StreamWriter` would break code.

A ringbuf (though a very good idea) won't make this problem go away as the passed data may be larger than the ringbuf.

robert-hh, 2022-09-10:
From the documentation it looks right for `uart.write` to return a number less than the message size. It's may be intersting if 0 is accepted, meaning that at the moment nothing can be written. In that case the first easy change for the drivers would be to return 0 when not data can be sent w/o blocking. timeout could be used for an estimation, how much data can be delivered for sending in that time without blocking.

robert-hh, 2022-09-10:
So I checked it with the mimxrt port, let `uart.write` return 0 if the timeout is less than the expected blocking time. There is no error raised, just 0 returned. Is that something `StreamWriter` can hanlde?

peterhinch, 2022-09-11:
There are two components to the `uasyncio` interface: the `.ioctl` in `machine_uart.c` and the `.write` method in `uart.c`. The `.ioctl` should (when queried) set `MP_STREAM_POLL_WR` if the UART is capable of accepting at least one byte of data. This will normally mean that `.write` never returns 0 as `.ioctl` is polled before `.write` is issued.

dpgeorge, 2022-09-11:
There is no output buffering of UART on stm32, so that's the reason it blocks when data is TX'd on the UART.

There isn't really a spec for how `machine.UART` should behave exactly, but blocking for a long time in an asyncio app is not good, so would be good to fix it.

Considering I2C and SPI, they also block, but I guess with those there is no expectation that they are non-blocking, because they have no timeout parameter like the UART does.  A UART is very much like a stream (eg TCP socket) so it's non-blocking behaviour should match a stream.

To fix it would require either sending only 1 byte for each call to `UART.write`, or by implementing a buffering mechanism (possibly with the help of DMA).  I guess a buffering mechanism makes the most sense and matches other ports like esp32 and rp2.  It could be made configurable (using the `txbuf` arg).  If the UART is in a non-blocking mode and there's no room left in the buffer then `UART.write` should return (once it fills the buffer up).

robert-hh, 2022-09-11:
>  If the UART is in a non-blocking mode and there's no room left in the buffer then UART.write should return (once it fills the buffer up).

The rp2 implementation is close to that. Only for timeout it uses timeout_char, which is forced to one character send time as the smallest value. As a consequence it will not timeout as long as there is no flow control stopping the send process. To get a timeout, either the check must be made against timeout, or timeout_char is allowed to be 0.

robert-hh, 2022-09-11:
So I changed machine_uart.c of the rp2 port such that `machine_uart_write()` will return the number of bytes delivered without waiting longer than the configured `timeout`. Then, if I call uart.write() with a 400 byte message and txbuf=256, timeout=0, it will return a number like 289, depending on the baud rate and timeout settings.

peterhinch, 2022-09-12:
@robert-hh That sounds good to me.

dpgeorge, 2022-09-12:
> The unbuffered solution would surely reduce `StreamWriter` throughput at high baudrates.

Yes.  We should implement a TX buffer on stm32.

robert-hh, 2022-09-12:
I could try to do so, but the STM32 code has gone through many iterations, and there is always the coexistence of the pyb and machine classes. So it will take a while.

robert-hh, 2022-09-13:
>  I guess a buffering mechanism makes the most sense and matches other ports like esp32 ...

ESP32 UART buffers, but blocks if the data size is larger than the buffer size, for  about the time until the excess characters are sent. e.g. for txbuf=256, data size=300, it will block for ~50 characters. This is different to what the API documentation tells.