micropython/micropython #9236: ESP32 Core panic (LoadProhibited) on wlan.config()

paullaffitte, 2022-09-07:
I'm experiencing core panic after upgrading from [v1.18](https://micropython.org/resources/firmware/esp32-20220117-v1.18.bin) to [v1.9.1](https://micropython.org/resources/firmware/esp32-20220618-v1.19.1.bin) on an ESP32 chip. Here are the logs:

```
*** pulse DTR ***
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
/home/micropython/esp-idf-v4.2/components/freertos/queue.c:7Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000127a  PS      : 0x00060130  A0      : 0x8013efee  A1      : 0x3ffccf80  
A2      : 0x3f41dbbc  A3      : 0x00000000  A4      : 0x3ffce2ac  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x3ffd5ba4  A8      : 0x00000057  A9      : 0x3ffccf30  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffd5be0  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3ffd5bb8  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace:0x40001277:0x3ffccf80 0x4013efeb:0x3ffccf90 0x4013fc55:0x3ffccfb0 0x4013e7ef:0x3ffccfe0 0x4013ec40:0x3ffcd010 0x400f94c2:0x3ffcd030 0x400dbdc5:0x3ffcd0f0 0x400e2679:0x3ffcd120 0x400e27a9:0x3ffcd140 0x40084789:0x3ffcd160 0x400dbd66:0x3ffcd200 0x400e2679:0x3ffcd260 0x400e26a2:0x3ffcd280 0x400efbca:0x3ffcd2a0 0x400eff71:0x3ffcd330 0x400d4cea:0x3ffcd350

ELF file SHA256: 00495403985fb7e0

Rebooting...
```

Here is the most minimal example I could write that demonstrate the bug:

```py
import network

hostname = 'esp32'
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.config(dhcp_hostname=hostname)
```

What is funny is that if I add almost anything between `wlan.active` and `wlan.config`, it "fixes" the issue. It can be a print statement, a small loop, some calculations... It's like the wlan controller is being initialized with `wlan.active` but doesn't have enough time to setup itself before `wlan.config` get called. So the following interestingly **works**:

```py
import network

hostname = 'esp32'
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
print('fix')
wlan.config(dhcp_hostname=hostname)
```

RSC-Games, 2022-09-09:
Hello paullaffitte,

I decided to test the issue because I also have mpy 1.19.1 installed on my ESP32. However, it is a custom build and results may be slightly different. Also, my 1.19 firmware is compiled with IDF 4.4, but my 1.18 firmware was built with 4.2. To my understanding, this was not an issue in 1.18.

I got this output:
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40093e30  PS      : 0x00060233  A0      : 0x80090c01  A1      : 0x3ffc4040  
A2      : 0x00000054  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x0000cdcd  
A6      : 0x00060223  A7      : 0x00060220  A8      : 0x0000cdcd  A9      : 0xffffffff  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x0000cdcd  
A14     : 0x003fffff  A15     : 0x00060223  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000054  LBEG    : 0x4008d9d2  LEND    : 0x4008d9dd  LCOUNT  : 0x00000000  

Backtrace:0x40093e2d:0x3ffc40400x40090bfe:0x3ffc4080 0x401384ae:0x3ffc40c0 0x40138511:0x3ffc40e0 0x401394fa:0x3ffc4100 0x40137d3d:0x3ffc4130 0x40137d5e:0x3ffc4160 0x401c5edd:0x3ffc4180 0x401c5fac:0x3ffc41e0 
```

I seem to recall that in the documentation, or in some other source, when bringing the WLAN up, you should poll the WLAN module with

```
while not wlan.active():
    pass
```
and that should hopefully prevent the error. This may be a bug of some sort, but I'm not sure it is related to MicroPython, as MicroPython runs on Core 1 on my device and Core 0 crashed. If anything, if this is important enough, we may want to open a bug on esp-idf.

paullaffitte, 2022-09-12:
Hello,

Thanks for your answer. I've tried with the while loop you advised me to add, but it seems that it doesn't even enter the loop (I've added a print statement inside too) but still crashes.

About the backtrace, how should I decode it ? Is there a tool to do so or I need to recompile mpy with debug flags ?

RSC-Games, 2022-09-12:
Huh. 

To decode your backtrace, you need a tool like idf-monitor so if you don't have esp-idf, that may not be feasible. I really need to open an enhancement issue that allows you to decode your backtrace online if you are using stock firmware. I feel like that would help a lot of people.

For my test I did, I shut down the WLAN unit and then re-initialized it (because my pycode bootloader starts the wlan unit). So this issue not only pertains to initially starting the driver but also restarting it. I retested the unit. On all tests, the payload in injected via the raw repl.

Your minimal example (causes crash) with my bootloader:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
~ snip ~
I (1695) init: Printing specifications.
I (1699) init: Generation:              1
I (1703) init: Revision:                1
I (1707) init: Firmware Version:        210
I (1711) init: Dev. Mode Flag:          0
I (1715) init: USB Debug Enabled:       0
I (1719) init: Finished printing hardware descriptor list.
I (1725) init: Printing System specifications.
I (1729) init: CPU:                     Dual-Core Xtensa LX6 @ 240 MHz
I (1735) init: RAM:                     4 MB SPIRAM
I (1739) init: Flash FS:                2 MB
I (1741) init: End of System specifications.
I (2071) init_mnt: Checked filesystem on SPI_FLASH successfully.
I (2072) init: SD Card detected.
I (2073) init_mnt: Preparing SD Card for mount.
I (2096) init_mnt: Checked SD Card filesystem successfully.
I (2150) init_mnt: Mounted SPI_FLASH successfully. It is accessible at "/".
I (2170) init_mnt: Mounted SD Card successfully. It is accessible at "/sd".
~ snip ~
I (4028) wifi:wifi driver task: 3ffc647c, prio:23, stack:6656, core=0
I (9160) system_api: Base MAC address is not set
I (9160) system_api: read default base MAC address from EFUSE
I (9178) wifi:wifi firmware version: 7679c42
I (9178) wifi:wifi certification version: v7.0
I (9178) wifi:config NVS flash: enabled
I (9180) wifi:config nano formating: disabled
I (9184) wifi:Init data frame dynamic rx buffer num: 32
I (9188) wifi:Init management frame dynamic rx buffer num: 32
I (9194) wifi:Init management short buffer num: 32
I (9198) wifi:Init dynamic tx buffer num: 32
I (9202) wifi:Init tx cache buffer num: 32
I (9206) wifi:Init static rx buffer size: 1600
I (9210) wifi:Init static rx buffer num: 10
I (9214) wifi:Init dynamic rx buffer num: 32
I (9218) wifi_init: rx ba win: 6
I (9222) wifi_init: tcpip mbox: 32
I (9226) wifi_init: udp mbox: 6
I (9230) wifi_init: tcp mbox: 6
I (9234) wifi_init: tcp tx win: 5744
I (9238) wifi_init: tcp rx win: 5744
I (9242) wifi_init: tcp mss: 1440
Deprecation warning! menu.status_graphics management will be detached
from hal.hardware.init_hardware in firmware build 250.
<ctrl-c>
MicroPython v1.19.1-18-g6e868d47d-dirty on 2022-06-17; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> from hal import hardware
>>> hardware.wlan.active(False)
False
>>> 
raw REPL; CTRL-B to exit
>OKGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008ddba  PS      : 0x00060130  A0      : 0x80138653  A1      : 0x3ffc24d0  
[... 95 lines cut ...]
    
    # Start the access point.
    wlan.active(True)
    wlan.config(essid=ssid, authmode=network.AUTH_WPA_WPA2_PSK, password=psk)
    
    # Wait for WiFi hardware to finish initializing.
    # WORKAROUND for https://github.com/micropython/micropython/issues/9236
    # Weird how this is placed after the above code without consequence.
    while wlan.active() != True:
        pass
```

The above code works, but the system is in AP mode. Maybe dhcp_hostname is not meant to be used in STA mode? This should be checked for by the MicroPython firmware.

This code appears to fix it.
```
import network

hostname = 'esp32'
wlan = network.WLAN(network.AP_IF)  # WAS STA_IF
wlan.active(True)
while not wlan.active(True):
    print("wait wlan")
wlan.config(dhcp_hostname=hostname)
```

RSC-Games, 2022-09-12:
It would appear this was also mentioned in #8792.

dpgeorge, 2023-01-20:
This is hopefully fixed by d6bc34a13aa734d8b32e5768c021377ac4815029

closed 2023-01-20 by dpgeorge