micropython/micropython #9271: Thread and irq don't go

schamu, 2022-09-10:
* Please search existing issues before raising a new issue. For questions about MicroPython or for help using MicroPython, or any sort of "how do I?" requests, please use the Discussions tab or raise a documentation request instead.

* In your issue, please include a clear and concise description of what the bug is, the expected output, and how to replicate it.

* If this issue involves external hardware, please include links to relevant datasheets and schematics.

* If you are seeing code being executed incorrectly, please provide a minimal example and expected output (e.g. comparison to CPython).

* For build issues, please include full details of your environment, compiler versions, command lines, and build output.

* Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name
 - git commit hash and port/board
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.

robert-hh, 2022-09-10:
Please use three backticks to enclose code in your post. You have only two.

schamu, 2022-09-10:
It's ma error
please delete

schamu, 2022-09-10:
it is my error, please delete

closed 2022-09-10 by schamu

robert-hh, 2022-09-10:
No problem. You can delete it yourself, but the history remains. But it also my experience that writing up a problem for others to read helps a lot to find a solution. And it is important that this text is actually sent. In that respect, it is good that you have sent it