micropython/micropython #9280: utime.sleep_us() overflow

socketpair, 2022-09-11:
```python
>>> from utime import sleep_us
>>> sleep_us(24*3600*1000_000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
```

It's not documented about any limits on the argument. Either fix in documentation or fix. Fixing is better.

RSC-Games, 2022-09-12:
It looks like you're trying to wait for 86.4 billion microseconds. Is there a reason why you are using time.sleep_us instead of time.sleep(24*3600)?

dpgeorge, 2022-09-12:
For efficiency, the sleep and ticks functions are restricted to machine-sized integers, eg 32-bit (signed) integers.

There are many, many places where this kind of error can be raised, for example:
```
>>> machine.Pin(10000000000000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
>>> 2<<10000000000000000000000
OverflowError: overflow converting long int to machine word
```
That last one even exists in CPython.

It's not feasible to fix these.  The best we could do is document it.

socketpair, 2022-09-12:
I use ESP32-S chip.

1. Why. The docs say that some boards may not handle floating point in `.sleep()`. I have `timedelta().total_seconds()` which is `float`. In order to make code portable, instead of `sleep(timedelta.total_seconds())` I decide to write `sleep_us(int(timedelta.total_seconds() * 1000_000))`
2. huh, `utime.sleep(24*3600)` does not work either! It sleeps only 500 seconds! Seems some kind of overflow. It's another nasty bug!

socketpair, 2022-09-12:
And actually, we CAN fix sleep functions. They need to split passed big value to portions and sleep multiple times. Something like:

```python
_limit = 1000 # just an example, need to say board specific real limit.
def sleep_us(value: int):
    while value > _limit:
        _real_sleep(limit)
        value -= limit
    _real_sleep(value) # zero is OK here, I think.
```

However it's not precise (may sleep slightly more).

jimmo, 2022-09-12:
Just FYI, do not use sleep_us for long sleeps as it is a spin sleep on most ports (i.e. uses lots of power). It's designed for only very short intervals where you need precise sleep time. Prefer `time.sleep(seconds)` or better, `time.sleep_ms()` (to avoid floats).

socketpair, 2022-09-12:
@jimmo thank you. I did not know. PLEASE add it to the documentation.

dpgeorge, 2022-09-12:
> And actually, we CAN fix sleep functions. They need to split passed big value to portions and sleep multiple times. Something like:

Yes, this is a good way to provide a longer sleep-us, that you can implement in your own code.  It's not a good idea to add this complexity to the core implementation.

> PLEASE add it to the documentation.

The differences between sleep/sleep_ms/sleep_us are already documented.  Eg for sleep_ms it explains that interrupts can occur and for more precise timing use sleep_us:

```
   This function will delay for at least the given number of milliseconds, but
   may take longer than that if other processing must take place, for example
   interrupt handlers or other threads.  Passing in 0 for *ms* will still allow
   this other processing to occur.  Use `sleep_us()` for more precise delays.
```

And note that sleep_us is telling you that it doesn't support long sleeps (by raising the exception).  That's much better than just silently sleeping for a shorter time.

> `utime.sleep(24*3600)` does not work either! It sleeps only 500 seconds

If you think you found a bug with this then please open a separate ticked for that, with details of hardware, MicroPython version, and exactly how to reproduce.  Then we can try to fix it.

socketpair, 2022-09-12:
Okay. So, please add to documentation:
1. limits on value that can be passed to sleep_us
2. information that some boards it may be implemented using busyloop which is not power-consumption friendly comparing to sleep_ms() and sleep().

> already documented.

No, the connection between precision and presence of busyloop is not obvious. For example, Linux's `nanosleep()` is not implemented using busyloop. I came from Linux world, so some things are not expected for me (and I guess for others).