micropython/micropython #9281: Hi!

sgyuri2003, 2022-09-12:
* Please search existing issues before raising a new issue. For questions about MicroPython or for help using MicroPython, or any sort of "how do I?" requests, please use the Discussions tab or raise a documentation request instead.

* In your issue, please include a clear and concise description of what the bug is, the expected output, and how to replicate it.

* If this issue involves external hardware, please include links to relevant datasheets and schematics.

* If you are seeing code being executed incorrectly, please provide a minimal example and expected output (e.g. comparison to CPython).

* For build issues, please include full details of your environment, compiler versions, command lines, and build output.

* Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name
 - git commit hash and port/board
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.

GeorgeCGV, 2022-09-28:
Tried to build Zephyr port

```
west build -b nucleo_h743zi -s modules/micropython/ports/zephyr -- -DCONF_FILE=prj_minimal.conf
```

result is a similar error:

```
<redacted> | sed "s/^\\\"\(Q(.*)\)\\\"/\1/" > <redacted>/build/genhdr/qstrdefs.preprocessed.h"

sed: -e expression #1, char 10: unterminated `s' command
cat: write error: No space left on device
'\"/"' is not recognized as an internal or external command,
operable program or batch file.

ninja: build stopped: subcommand failed.

FATAL ERROR: command exited with status 1: 'C:\cmake\bin\cmake.EXE' --build '<redacted>\build'
```

GeorgeCGV, 2022-09-28:
@sgyuri2003 could you test if #9451 fixes the issue for you?