micropython/micropython #9282: ESP32 build error

sgyuri2003, 2022-09-12:
Windows 
ESP-IDF 4.4 cmd 
idf.py -D MICROPY_BOARD=GENERIC_SPIRAM build

[1153/1380] Generating ../../genhdr/qstrdefs.preprocessed.h
FAILED: genhdr/qstrdefs.preprocessed.h C:/micropython/ports/esp32/build/genhdr/qstrdefs.preprocessed.h
esp-idf\main\CMakeFiles\qstrdefs.preprocessed.h-0fb6a53.bat 7fb35ac92d7b64e3
cat: C:/micropython/ports/esp32/build/genhdr/qstrdefs.collected.h: No such file or directory
sed: -e expression #1, char 10: unterminated `s' command
'\"/"' is not recognized as an internal or external command,
operable program or batch file.
[1154/1380] Generating ../../genhdr/moduledefs.h
FAILED: genhdr/moduledefs.h C:/micropython/ports/esp32/build/genhdr/moduledefs.h
cmd.exe /C "cd /D C:\micropython\ports\esp32\build\esp-idf\main && C:\Users\Admin\AppData\Local\Programs\Python\Python39\python3.exe C:/micropython/py/makemoduledefs.py C:/micropython/ports/esp32/build/genhdr/moduledefs.collected > C:/micropython/ports/esp32/build/genhdr/moduledefs.h"
Traceback (most recent call last):
  File "C:\micropython\py\makemoduledefs.py", line 82, in <module>
    main()
  File "C:\micropython\py\makemoduledefs.py", line 77, in main
    modules = find_module_registrations(args.file[0])
  File "C:\micropython\py\makemoduledefs.py", line 27, in find_module_registrations
    with io.open(filename, encoding="utf-8") as c_file_obj:
FileNotFoundError: [Errno 2] No such file or directory: 'C:/micropython/ports/esp32/build/genhdr/moduledefs.collected'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

closed 2022-09-12 by sgyuri2003

sambeard, 2025-01-02:
I have the same issue, presumably because i'm building on windows, any advice on how this was resolved?

sgyuri2003, 2025-01-03:
hi
0. install Python 3.x
windows advanced settings , add to path list python and python/script

1. https://dl.espressif.com/dl/esp-idf/?idf=4.4
[image: image.png]
2. Dowload ESP-IDF v3.3.2 offline installer
3 install esp IDF
4 windows advanced settings , add to path list idf.py path
5 click on the ESP-IDF icon that appears on the desktop
6 In the command window (which contains the ESP-IDF environment), set the
device type as usual.
7 From this command window, navigate to the appropriate directory for the
python sdk and you can start compiling.
8 windows advanced settings , add to path list idf.py path

Unfortunately, the recommendation may not be complete, because I put it
together from memory, I haven't dealt with it for a year, but only because
I had other things to do.
With friendly regards, sgyuri2003

Sam Beard ***@***.***> ezt írta (időpont: 2025. jan. 2., Cs,
14:18):

stinos, 2025-01-04:
> windows advanced settings , add to path list idf.py path

In general it's advisable to not set the global PATH variables. Rather use a batch file to temprorarily set this up or similar instead: allows easy switching between environments/versions/..., a lot less issuses with conflicting items in PATH etc.

And in this particular case because it should not be needed: esp-idf comes with a [batch file](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#using-the-command-prompt) which sets up the PATH etc correctly.

Same goes for the Python installation: all common installers (CPython/Miniconda/standalone portable CPython/...) already come with their own way to setup the PATH (and possibly PYTHONPATH etc) correctly.