micropython/micropython #9286: windows: os.sep should be backslash

jimmo, 2022-09-12:
This came up in https://github.com/micropython/micropython-lib/pull/527, and we have a workaround there, but getting this right will be required to make os.path work correctly.

The windows port current returns `/` from `os.sep` but should be `\`. We should fix that, as well as implementing `os.altsep` for unix/windows (return None on unix, `/` on windows).

dogtopus, 2022-09-14:
Even better: instead of/other than just have a Win32API specific VFS, make this configurable in mpconfigport for POSIX, so some cursed platforms that uses backslash as separator but provides POSIX-like filesystem API would work better out of the box (e.g. MinGW, a platform that I'm currently making homebrew for, and probably more because you never know what crazy things proprietary sw developer would do to satisfy insane requitements)

jimmo, 2022-09-15:
> make this configurable in mpconfigport for POSIX

@dogtopus Yes this is exactly how it would work and you'd be able to override it on any port (or board / variant).