micropython/micropython #9304: utime.sleep(24*3600) sleeps only 500 seconds on ESP32

socketpair, 2022-09-14:
boot.py:

```python
from machine import freq
from utime import sleep, time

    freq(240000000)
    while True:
        print(time())
        sleep(24 * 3600)
```

will print a line roughly at every 500 seconds:

```
 1
501
1002
...
```

socketpair, 2022-09-14:
Sleeping for `24 * 3600 - 250` is expectedly sleeps for 50 seconds. So, I guess, it's overflow+wrap-around issue.

socketpair, 2022-09-14:
`sleep((24 * 3600 - 500)//5)` competes immediately

dpgeorge, 2022-09-14:
Thanks for the report, I can confirm the issue on the latest master build.

It looks like an overflow of a 32-bit integer when computing microseconds, it should be a uint64_t calculation.

dpgeorge, 2022-09-14:
Should be fixed by 0e8c2204da377e95b5000a3c708891d98cdeb69c

closed 2022-09-14 by dpgeorge

socketpair, 2022-09-14:
@dpgeorge I can test the change, but I need some way of OTA update of Micropython. is it possible? If yes, some link to documentation will greatly help.

dpgeorge, 2022-09-14:
It is possible to do OTA update of ESP32 firmware using `esp32.Partition` objects (see http://docs.micropython.org/en/latest/library/esp32.html?highlight=esp32#flash-partitions).  But for that to work you first need to prepare the esp32 to have the correct OTA partition layout, which really needs a physical USB connection.