micropython/micropython #9313: fat filesystem can rename a directory inside of itself

jepler, 2022-09-14:
Over in Circuitpython, we noticed that (via the web workflow, though it's equally possible in the repl) it was possible to rename a directory "inside of itself", making the filesystem corrupt.

I created a failing test for this, specifically against the fat filesystem implementation: https://github.com/micropython/micropython/compare/master...jepler:circuitpython:vfs-fat-island?expand=1#files_bucket

I don't know whether other filesystems like littlefs are affected, I didn't test it.

jepler, 2022-09-14:
@makermelissa tagging you on this, I only saw some of the discussion about this and you may have more info to contribute. Looks like you are working on a fix for CP at https://github.com/adafruit/circuitpython/pull/6883 as well.

dhalbert, 2022-09-14:
I think #4177 may cover this.

jepler, 2022-09-14:
Yeah, I failed to find it in my search, thanks

makermelissa, 2022-09-14:
Ah ok. I didn't realize it was happening in the REPL or with MicroPython too.

jonnor, 2024-09-29:
Duplicate of https://github.com/micropython/micropython/issues/4177

closed 2024-09-29 by jonnor