micropython/micropython #9314: ESP32: Viper code can't be called from frozen bytecode module.

DusKing1, 2022-09-14:
I've tried my best to search for related issues, but I really can't find any, so if anyone finds something similar to mine, please let me know and I'll check the relevant issue/PR firstly.

My problem is that I have a piece of viper code that works fine when imported as a user file (.py), but compiles if I freeze it to bytecode and include it in the compiled firmware, the firmware that comes out cannot call that piece of viper code normally, and it will give the following error warning:

```
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x401de5bc  PS      : 0x00060530  A0      : 0x800dbc7a  A1      : 0x3ffccc00
A2      : 0x3ffe5880  A3      : 0x00000001  A4      : 0x3ffea6f0  A5      : 0x3f410b0c
A6      : 0x3f419250  A7      : 0x3f41eb2c  A8      : 0x801de5b7  A9      : 0x3ffccbd0
A10     : 0x3f410b0c  A11     : 0x00000005  A12     : 0x000000c8  A13     : 0x3ffccd54
A14     : 0x00000000  A15     : 0x3ffe5794  SAR     : 0x00000004  EXCCAUSE: 0x00000003
EXCVADDR: 0x3f410b0c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x401de5b9:0x3ffccc00 0x400dbc77:0x3ffccc40 0x400e2641:0x3ffccc60 0x400e2771:0x3ffccc80 0x40084749:0x3ffccca0 0x400dbd2e:0x3ffccd40 0x400e2641:0x3ffccda0 0x400e2771:0x3ffccdc0 0x40084749:0x3ffccde0 0x400dbd2e:0x3ffcce80 0x400e2641:0x3ffcceb0 0x400e266a:0x3ffcced0 0x400ff33a:0x3ffccef0 0x400ff3ad:0x3ffccf80 0x400ff66c:0x3ffccfb0 0x400e27ec:0x3ffcd090 0x400849a9:0x3ffcd0d0 0x400dbd2e:0x3ffcd170 0x400e2641:0x3ffcd1c0 0x400e266a:0x3ffcd1e0 0x400efb92:0x3ffcd200 0x400eff39:0x3ffcd290 0x400d4cca:0x3ffcd2b0

ELF file SHA256: 001762e7fe9dfd58
```

jimmo, 2022-09-14:
@DusKing1 Can you provide a repro for us? Does it happen with any viper function, or only your specific one?

DusKing1, 2022-09-14:
> @DusKing1 Can you provide a repro for us? Does it happen with any viper function, or only your specific one?

Hey thanks for your quick reply, feel so much better now!

The repo is currently private, but I'm happen to DM you if possible.

My viper code is just:
```
class CRSF_RC_Generator:
    def __init__(self):
        # print(str(time.ticks_us()) + " [ Init ] CRSF Generator")
        self.packet_buf = b'00000000000000000000000000'
        self._crc_tab = bytes([
                0x00,0xD5,0x7F,0xAA,0xFE,0x2B,0x81,0x54,0x29,0xFC,0x56,0x83,0xD7,0x02,0xA8,0x7D,
                0x52,0x87,0x2D,0xF8,0xAC,0x79,0xD3,0x06,0x7B,0xAE,0x04,0xD1,0x85,0x50,0xFA,0x2F,
                0xA4,0x71,0xDB,0x0E,0x5A,0x8F,0x25,0xF0,0x8D,0x58,0xF2,0x27,0x73,0xA6,0x0C,0xD9,
                0xF6,0x23,0x89,0x5C,0x08,0xDD,0x77,0xA2,0xDF,0x0A,0xA0,0x75,0x21,0xF4,0x5E,0x8B,
                0x9D,0x48,0xE2,0x37,0x63,0xB6,0x1C,0xC9,0xB4,0x61,0xCB,0x1E,0x4A,0x9F,0x35,0xE0,
                0xCF,0x1A,0xB0,0x65,0x31,0xE4,0x4E,0x9B,0xE6,0x33,0x99,0x4C,0x18,0xCD,0x67,0xB2,
                0x39,0xEC,0x46,0x93,0xC7,0x12,0xB8,0x6D,0x10,0xC5,0x6F,0xBA,0xEE,0x3B,0x91,0x44,
                0x6B,0xBE,0x14,0xC1,0x95,0x40,0xEA,0x3F,0x42,0x97,0x3D,0xE8,0xBC,0x69,0xC3,0x16,
                0xEF,0x3A,0x90,0x45,0x11,0xC4,0x6E,0xBB,0xC6,0x13,0xB9,0x6C,0x38,0xED,0x47,0x92,
                0xBD,0x68,0xC2,0x17,0x43,0x96,0x3C,0xE9,0x94,0x41,0xEB,0x3E,0x6A,0xBF,0x15,0xC0,
                0x4B,0x9E,0x34,0xE1,0xB5,0x60,0xCA,0x1F,0x62,0xB7,0x1D,0xC8,0x9C,0x49,0xE3,0x36,
                0x19,0xCC,0x66,0xB3,0xE7,0x32,0x98,0x4D,0x30,0xE5,0x4F,0x9A,0xCE,0x1B,0xB1,0x64,
                0x72,0xA7,0x0D,0xD8,0x8C,0x59,0xF3,0x26,0x5B,0x8E,0x24,0xF1,0xA5,0x70,0xDA,0x0F,
                0x20,0xF5,0x5F,0x8A,0xDE,0x0B,0xA1,0x74,0x09,0xDC,0x76,0xA3,0xF7,0x22,0x88,0x5D,
                0xD6,0x03,0xA9,0x7C,0x28,0xFD,0x57,0x82,0xFF,0x2A,0x80,0x55,0x01,0xD4,0x7E,0xAB,
                0x84,0x51,0xFB,0x2E,0x7A,0xAF,0x05,0xD0,0xAD,0x78,0xD2,0x07,0x53,0x86,0x2C,0xF9
            ])
    def build_rc_packet_new2(self, chx) -> bytes:
        self.chx = chx
        self._append_bits()
        return self.packet_buf
    @micropython.viper
    def _append_bits(self) -> int:
        buf = ptr8(self.packet_buf)
        
        buf[0] = 0xC8
        buf[1] = 0x18
        buf[2] = 0x16
        buf[3] = int((int(self.chx[0])& 0x07FF))
        buf[4] = int((int(self.chx[0])& 0x07FF)>>8 | (int(self.chx[1]) & 0x07FF)<<3)
        buf[5] = int((int(self.chx[1])& 0x07FF)>>5 | (int(self.chx[2]) & 0x07FF)<<6)
        buf[6] = int((int(self.chx[2])& 0x07FF)>>2)
        buf[7] = int((int(self.chx[2])& 0x07FF)>>10 | (int(self.chx[3]) & 0x07FF)<<1)
        buf[8] = int((int(self.chx[3])& 0x07FF)>>7 | (int(self.chx[4]) & 0x07FF)<<4)
        buf[9] = int((int(self.chx[4])& 0x07FF)>>4 | (int(self.chx[5]) & 0x07FF)<<7)
        buf[10] = int((int(self.chx[5])& 0x07FF)>>1)
        buf[11] = int((int(self.chx[5])& 0x07FF)>>9 | (int(self.chx[6]) & 0x07FF)<<2)
        buf[12] = int((int(self.chx[6])& 0x07FF)>>6 | (int(self.chx[7]) & 0x07FF)<<5)
        buf[13] = int((int(self.chx[7])& 0x07FF)>>3)
        buf[14] = int(int(self.chx[8]) & 0x07FF)
        buf[15] = int((int(self.chx[8])& 0x07FF)>>8 | (int(self.chx[9]) & 0x07FF)<<3)
        buf[16] = int((int(self.chx[9])& 0x07FF)>>5 | (int(self.chx[10]) & 0x07FF)<<6)
        buf[17] = int((int(self.chx[10])& 0x07FF)>>2)
        buf[18] = int((int(self.chx[10])& 0x07FF)>>10 | (int(self.chx[11]) & 0x07FF)<<1)
        buf[19] = int((int(self.chx[11])& 0x07FF)>>7 | (int(self.chx[12]) & 0x07FF)<<4)
        buf[20] = int((int(self.chx[12])& 0x07FF)>>4 | (int(self.chx[13]) & 0x07FF)<<7)
        buf[21] = int((int(self.chx[13])& 0x07FF)>>1)
        buf[22] = int((int(self.chx[13])& 0x07FF)>>9 | (int(self.chx[14]) & 0x07FF)<<2)
        buf[23] = int((int(self.chx[14])& 0x07FF)>>6 | (int(self.chx[15]) & 0x07FF)<<5)
        buf[24] = int((int(self.chx[15])& 0x07FF)>>3)
        buf[25] = int(self._calc_crc(buf))
        return 0

    @micropython.viper
    def _calc_crc(self, buf: ptr8) -> int:
        crc = 0x00
        index = 0x00
        crc_tab = ptr8(self._crc_tab)
        for i in range(2,25):
            crc = crc_tab[int(crc ^ buf[i])]
        return crc
```

jimmo, 2022-09-14:
`self.packet_buf = b'00000000000000000000000000'`

When you freeze your .mpy this ends up in ROM. You code is modifying this buffer. Replace with

`self.packet_buf = bytearray(16)`

closed 2022-09-14 by jimmo

DusKing1, 2022-09-14:
oh, wow, thanks!!