micropython/micropython #9316: pico - zombie pio state machines after soft reset

ccanepa, 2022-09-14:
version:
```python
>>> import sys
>>> sys.implementation
(name='micropython', version=(1, 19, 1), _machine='Raspberry Pi Pico with RP2040', _mpy=4102)
```

Problem description: pio sm instances  can be unexpectedly resucitated after a soft reset

More detailed description:
  - instantiate and activate many pio state machines with the same pio program
  - call machine.soft_reset()
  - instantiate and activate only one sm, with the same pio program
  - Expected: only one sm activelly driven his output
  - Observed: all sm with same pio program will be activelly driven their outputs

The demo script expects to have led + resistor in gpios 10..13

It will instantiate and activate four sm, each blinling a led, do a soft reset, instantiate and activate only one sm, do a hard reset and repeat. 

Expected 5 seconds of four led blinking, followed by 5 seconds of only one led blinking, repeat

Observed always four leds blinking

See docstring for instuctions to run

demo script: pio_and_machine_reset.py
```python
""" demoing zombie state machines after soft reset

Using a pico lean and mean, not W or other variants
HW: led + resistor on gpios 10..13

Shoul do: blink the four leds 5 seconds, then blink only the led in gpio 13
Observed: four leds blinking always

Operation:
   - if not first run, ensure the file 'soft_reset_flag.txt' does not
     exist in the pico
   - Close Thonny or other runner / debugger to discard other sw interference 
   - Unplug and plug the pico to ensure a hard reset
   - Use minicom or Bootterm ( bt ) to get the serial output from the pico
"""

from machine import Pin
from rp2 import PIO, StateMachine, asm_pio
import time

def report_state_machines():
    for p in (0, 1):
        for sm_c in range(4):
            fmt = "rp2.PIO(%d).state_machine(%d).active(): %s"
            print( fmt % (p, sm_c, rp2.PIO(p).state_machine(sm_c).active()))

class FSFlag:
    def __init__(self, name):
        self.name = name
        self.fname = name + "_flag.txt"
        
    def is_flag_set(self):
        present = False
        try:
            f = open(self.fname, "r")
            present = True
            f.close()
        except Exception:
            present = False
        return present

    def set_(self):
        f = open(self.fname, "w")
        f.write("flag file")
        f.close()

    def clear(self):
        import os
        if self.is_flag_set():
            os.remove(self.fname)

    def value(self, val=None):
        if val is None:
            return self.is_flag_set()
        elif val:
            self.set_()
        else:
            self.clear()

@rp2.asm_pio(set_init=rp2.PIO.OUT_LOW)
def blink():
    set(pins, 1)   [31]
    nop()          [31]
    nop()          [31]
    nop()          [31]
    nop()          [31]
    set(pins, 0)   [31]
    nop()          [31]
    nop()          [31]
    nop()          [31]
    nop()          [31]

def blink_1_led(dt):
    print("only led on gpio 13 should blink")
    print("only SM 3 should be active")
    sm = StateMachine(3, blink, freq=2000, set_base=Pin(13, Pin.OUT, value=0))
    sm.active(1)
    report_state_machines()
    time.sleep(dt)

def blink_4_leds(dt):
    print("leds in gpios (10, 11, 12, 13) should blink")
    print("SMs 0..4 should be active")
    gpios = (10, 11, 12, 13)
    py_sm_ids = (0, 1, 2, 3)
    pairs = [(id_, Pin(gp, Pin.OUT, value=0)) for id_, gp in zip(py_sm_ids, gpios)]
    sms = {id_: StateMachine(id_, blink, freq=2000, set_base=pin) for id_, pin in pairs} 
    for sm in sms.values():
        sm.active(1)
    report_state_machines()
    time.sleep(dt)

print("Comming Alive")
report_state_machines()
soft_reset = FSFlag("soft_reset")
if soft_reset.value():
    print("after soft reset")
    blink_1_led(5)
    soft_reset.value(0)
    machine.reset()
else:
    print("after hard reset")
    blink_4_leds(5)
    soft_reset.value(1)
    machine.soft_reset()
```

Additional info, not demoed but can easily show with variants on the demo code:
- always after soft reset all the sm active before the reset are shown as active by  `rp2.PIO(p).state_machine(sm_c).active()`
- if no sm is instantiated after the soft reset, no pin activity shows in the pins controlled by the old sm's
- if a pio sm with a __different__ pio program is instantiated after the soft reset,  no pin activity shows in the pins controlled by the old sm's
- if a pio sm with __same__ pio program is activated after the soft reset, the old sm's activelly drive their pins

This issue is probably related to
https://forums.raspberrypi.com/viewtopic.php?t=340099
https://github.com/thonny/thonny/issues/2455

iabdalkader, 2022-09-25:
It might be because state machines are not disabled in `rp2_pio_deinit`, they might be disabled/reset somewhere else that I missed, but worth checking if calling `sm.active(0)` on every state machine, before the soft-reset fixes the issue.

ccanepa, 2022-09-25:
Yes, changing the bugdemo to deactivate all sm before soft_reset fixes the two wrong behaviours demo-ed in the demo.

Still, the PIO does not go to a known state after that fix: adding in function blink_1_led from bugdemo, after sm.active(1), the line 
`rp2.PIO(0).state_machine(0).active(1)`
the led tied to the gpio pin controlled by former sm0 will behave as still controlled by the old sm0

Then it seems that the soft reset clears the PIO program memory but do not change other PIO registers,

But this is more in the gray area about what a soft_reset should do to PIO state, and more in general, what each reset cause promises about the PIO state?

iabdalkader, 2022-09-26:
I can't follow your example it's too complex, but I managed to reproduce this with the simple pio 1hz example. I did the following:
* sm0 -> toggle pin 1hz
* sm1 -> toggle LED 1hz
soft-reset (via ctrl-D)
* sm1 -> toggle LED 1hz 
* Both LED and pin are toggling.

ccanepa, 2022-09-26:
Yeah, that will be equivalent to show unexpected PIO sm activity after a soft_reset.

If you print all the sm active status after the soft reset you can also see that both sm0 and sm1 show as active, even before instantiating sm1 after the reset

Adding  into  `rp2_pio_deinit` 'clear all sm active bits' would be a good fix for the pico-non W, it is clearing the PIO memory so no reason to let them active. 

The problem would be for the pico-W or other boards that don't want to stop in a soft reset. Maybe the .uf2 for those boards should have a 'sms_not_to_reset' array and then  `rp2_pio_deinit` will deactivate only the sm not in the array.