micropython/micropython #9339: Rename SPI/I2C pins to OSHWA standard.

jimmo, 2022-09-17:
I missed this in January but the OSHWA published their guidelines for the new pins which was in the works for a couple of years -- https://www.oshwa.org/a-resolution-to-redefine-spi-signal-names/

I'm already starting to see these new names on silkscreens and documentation.

Supporting this at the kwarg-names and documentation level should be very straightforward, but we'll obviously need to maintain backwards compatibility with the old names. We already have precedent for doing this e.g. with the renamed wifi config kwargs like `dhcp_hostname` and `security`.

robert-hh, 2022-09-17:
We had that discussion earlier. Maybe I kicked it off myself. As far as I recall, we preferred SDI, SDO and SDIO. That's already used e.g. in the NXP data sheets. SAMD and STM still use MOSI and MISO.
Renaming the kwargs will be a mess.

jimmo, 2022-09-17:
Argh... I searched for issues before raising this. But it was in a PR, and I even commented and linked to the same OSHWA doc... See: https://github.com/micropython/micropython/pull/7367

(I don't understand the OSHWA site, there are no dates on the pages, but the post saying they are releasing the resolution is dated Jan 2022 -- https://www.oshwa.org/2022/01/10/new-spi-terminology/).

> Renaming the kwargs will be a mess.

Even if we support both the old and new names?

robert-hh, 2022-09-17:
> and I even commented and linked ....

You're getting old ... What should I say?

> Even if we support both the old and new names?

I assumed that, and still there will be inconsistencies.

sosi-deadeye, 2022-09-18:
You are making extra work for yourself because of PC. Nothing is improved, and we still have to remember the new names.

What comes next? Finding other politically incorrect words in the source code and documentation? Gender correct documentation? Support programs for women and LBGTQ+?

jimmo, 2022-09-18:
@sosi-deadeye This is not helpful or welcome here. Please see https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md

This is simple -- the names are changing and we should (also) support the new names.