micropython/micropython #9346: PicoW, MicroPython and textwrap, does it work fully? Or is it the re module?

brianlinuxing, 2022-09-17:
Hello,

Is the textwrap module in the library usable on MicroPython 1.19.1 nightly?

This error comes up when loading up textwrap.py of Micropython-3.4.2.post1

Line 83:

**Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
  File "<stdin>", line 83, in TextWrapper
ValueError: error in regex**

I wonder if there is a problem with the re package? Shown as **“        r'(\s+|'                                  # any whitespace”**  on line 83 as can be seen from the screenshots.

My question, is there an obvious debug strategy to this or short-term fix?

I found a previous open issue from 2018:

https://github.com/micropython/micropython-lib/issues/318

I would welcome helpful suggestions as to how to resolve this (a module I am using needs textwrap to work).

Thanks, Brian

[Running in Thonny 4.0.1 and MicroPython v1.19.1 on 2022-09-14; Raspberry Pi Pico W with RP2040]
[image]

jimmo, 2022-09-19:
Closing as a duplicate of https://github.com/micropython/micropython-lib/issues/318

For other readers, here's the reply I sent to @brianlinuxing on Discord:

```
fyi, the context here is that textwrap was likely written with the PCRE-based re module from micropython-lib for unix-ffi in mind, so there wouldn't have been any consideration for whether the built-in module supported it

(micropython-lib's history is very much based on the unix-ffi use case. when I tried to split out the modules that work standalone i likely missed the subtlety of this regexp)
```

We might one day add the necessary support to the built-in `re` module, but in the meantime it would be better to just workaround in the `textwrap` module.

closed 2022-09-19 by jimmo

brianlinuxing, 2022-09-19:
Indeed, my very first idea was to work around the issue.

I tried this approach:

1. With reference to the original library https://github.com/python/cpython/blob/main/Lib/textwrap.py
2. Changing the format of the statement.
3. Looking for alternatives (across GitHub, Google, etc)

At the moment, there are none.

I have spent hours looking and fiddling with the code to no avail.

Therefore, I am in a pickle, because

a) a module **I have to use** for my IoT app requires textwrap.
b) and the library which supplies it is broken, not faulty, but properly broken.

Given this is an **officially released library** which users might, not unreasonably, expect to run/compile or work etc, then I am at a loss as to a productive path to follow to either: work around it or fix it.

Equally, I am surprised the issue has been closed, when clearly the module is still broken.

The problem is localized to line 83:

wordsep_re = re.compile(
r"(\s+|" # any whitespace
r"[^\s\w]*\w+[^0-9\W]-(?=\w+[^0-9\W])|" # hyphenated words
r"(?<=[\w\!\"\'\&\.\,\?])-{2,}(?=\w))"
) # em-dash

If that can be fixed then all should be good, and I can proceed with my IoT solution. Thank you :)

jimmo, 2022-09-19:
> Equally, I am surprised the issue has been closed, when clearly the module is still broken.

This issue is closed because the bug is with something that is not in this repository. We are unlikely to add lookahead assertions to the built-in `re` library.

https://github.com/micropython/micropython-lib/issues/318 is the issue tracking this problem.

andrewleech, 2022-09-19:
> Therefore, I am in a pickle, because
> 
> a) a module **I have to use** for my IoT app requires textwrap. b) and the library which supplies it is broken, not faulty, but properly broken.

Please keep in mind that micropython is an open source project that is significantly built on community submissions. It wouldn't exist without the innovative and continuous efforts of @dpgeorge and the core team, but _many_ aspects are community contributions.

Textwrap in particular was contributed by a member of the community, some 8 years ago, without any claims of compatibility or explicit support. The git log shows it was copied verbatim from cpython at that time. 

Also it does work perfectly correctly, when used on the Unix port with the ffi based re module.

The embedded re module also works perfectly for the features it supports.

Nothing here is explicitly broken, its just unfortunate that textwrap currently requires features not offered be embedded re. 

Remember, this is an open source project. If you need it to have extra features right now, you're free and encouraged to add these features. I regularly need extra features for my professional work using micropython, it's amazing that I can add these, legally and with community support.

Please do contribute any improvements back if you do build them, that's what makes this community stronger.