micropython/micropython #9369: ESP32: BLE panic when receiving long NOTIFY messages

jonathanhogg, 2022-09-20:
I have an application that does software updates over BLE UART serial connections (my own implementation of BLE UART). This has recently started crashing when receiving ~400 byte messages. I push the MTU up to 515 at the beginning of the connection, so it's nothing to do with this.

Backtrace:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400daa7c  PS      : 0x00060230  A0      : 0x800dafc0  A1      : 0x3ffd5170  
0x400daa7c: gc_alloc at .../micropython/py/gc.c:583

A2      : 0x00000175  A3      : 0x00000000  A4      : 0x00060e20  A5      : 0x00000017  
A6      : 0x003fffff  A7      : 0x00060c23  A8      : 0x800daa7c  A9      : 0x3ffd5150  
A10     : 0x00000000  A11     : 0x3ffc43ac  A12     : 0x400ef924  A13     : 0x00000000  
0x400ef924: peripheral_gap_event_cb at .../micropython/extmod/nimble/modbluetooth_nimble.c:1178

A14     : 0x00000004  A15     : 0x00000003  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400e892c  LEND    : 0x400e8935  LCOUNT  : 0x00000000  
0x400e892c: mp_bluetooth_gattc_on_data_available at .../micropython/extmod/modbluetooth.c:1404 (discriminator 3)
0x400e8935: mp_bluetooth_gattc_on_data_available at .../micropython/extmod/modbluetooth.c:1403 (discriminator 3)

Backtrace:0x400daa79:0x3ffd51700x400dafbd:0x3ffd5190 0x400e893b:0x3ffd51b0 0x400ef341:0x3ffd5200 0x400ef67d:0x3ffd5240 0x400ef94d:0x3ffd5290 0x40116299:0x3ffd52f0 0x401162e3:0x3ffd5310 0x4011780a:0x3ffd5330 0x4011e2c9:0x3ffd5390 0x401133bd:0x3ffd53b0 0x4011328b:0x3ffd53e0 0x40112b34:0x3ffd5410 0x40112b3f:0x3ffd5430 0x4011b2fe:0x3ffd5450 0x400fa0df:0x3ffd5470 
0x400daa79: gc_alloc at .../micropython/py/gc.c:583
0x400dafbd: m_malloc at .../micropython/py/malloc.c:86
0x400e893b: mp_bluetooth_gattc_on_data_available at .../micropython/extmod/modbluetooth.c:1406
0x400ef341: gattc_on_data_available at .../micropython/extmod/nimble/modbluetooth_nimble.c:1292
0x400ef67d: commmon_gap_event_cb at .../micropython/extmod/nimble/modbluetooth_nimble.c:407
0x400ef94d: peripheral_gap_event_cb at .../micropython/extmod/nimble/modbluetooth_nimble.c:1208
 (inlined by) peripheral_gap_event_cb at .../micropython/extmod/nimble/modbluetooth_nimble.c:1178
0x40116299: ble_gap_call_event_cb at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:716
0x401162e3: ble_gap_call_conn_event_cb at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:742
0x4011780a: ble_gap_notify_rx_event at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:6110
0x4011e2c9: ble_att_svr_rx_notify at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2483
0x401133bd: ble_att_rx at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att.c:503
0x4011328b: ble_hs_hci_evt_acl_process at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:927
0x40112b34: ble_hs_process_rx_data_queue at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:238
0x40112b3f: ble_hs_event_rx_data at .../esp-idf-4.4/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532
0x4011b2fe: ble_npl_event_run at .../esp-idf-4.4/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at .../esp-idf-4.4/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78
0x400fa0df: ble_host_task at .../micropython/ports/esp32/mpnimbleport.c:43
```

It crashes on the very first long message that is received. If I reduce my block size down so that the messages are more like 200 bytes then the problem goes away.

Not sure of exact commit, but this was working a couple of months ago and is now exploding on 13c4470fd04981fce3fd29771e1d973febbe7d60. Application code hasn't changed in that time, just my build of MicroPython.

jonathanhogg, 2022-09-20:
Looking at the code, I assume that it is the fragmented buffer reassembly path in `mp_bluetooth_gattc_on_data_available()` that is failing, as the other path doesn't do any heap allocation.

I assume that this changed when the async BLE events came in. However, I'm certain that I've tested this code on a build after that as I had to remove the `.config(rxbuf=...)` in my code to get it to run again.

dpgeorge, 2022-09-21:
Thanks for the report.

When `MICROPY_PY_BLUETOOTH_USE_SYNC_EVENTS_WITH_INTERLOCK` the `m_new` and `m_del` calls in `mp_bluetooth_gattc_on_data_available()` need to be protected with the interlock...

dpgeorge, 2022-09-21:
Please see a suggested fix in #9379.  @jonathanhogg are you able to test that and see if it works?

jonathanhogg, 2022-09-21:
I get this instead, @dpgeorge:

```
Unhandled exception in IRQ callback handler
MemoryError: memory allocation failed, allocating 1073476631 bytes
```

which seems a somewhat unlikely allocation size. I can't see anything obviously wrong in the code at the moment, but I'll add some debug statements and see if I can figure out the bug…

jonathanhogg, 2022-09-21:
Printing out details of each fragment I get:

```
-- fragment 0 with 271 bytes @ 3ffd2e21
-- fragment 1 with 1073476360 bytes @ 3ffd2ce8
-- total 1073476631 bytes
Unhandled exception in IRQ callback handler
MemoryError: memory allocation failed, allocating 1073476631 bytes
```

That 1073476360 (0x3ffbf308) apparent length for the second fragment looks suspiciously like a memory address to me…

jonathanhogg, 2022-09-21:
Errr… this does not look safe/correct:

```c
void mp_bluetooth_gattc_on_data_available(uint8_t event, uint16_t conn_handle, uint16_t value_handle, const uint8_t **data, uint16_t *data_len, size_t num) {
    mp_int_t args[] = {conn_handle, value_handle};
    size_t data_len_size_t = *data_len;
    invoke_irq_handler(event, args, 2, 0, NULL_ADDR, NULL_UUID, data, &data_len_size_t, num);
}
```

Amending to:

```c
void mp_bluetooth_gattc_on_data_available(uint8_t event, uint16_t conn_handle, uint16_t value_handle, const uint8_t **data, uint16_t *data_len, size_t num) {
    mp_int_t args[] = {conn_handle, value_handle};
    size_t data_len_size_t[5];
    for (size_t i = 0; i < num; ++i) {
        data_len_size_t[i] = data_len[i];
    }
    invoke_irq_handler(event, args, 2, 0, NULL_ADDR, NULL_UUID, data, data_len_size_t, num);
}
```

I get sane behaviour:

```
-- fragment 0 with 271 bytes @ 3ffd2c75
-- fragment 1 with 87 bytes @ 3ffd2d94
-- total 358 bytes
-- reassembled fragments @ 3faa6030
```

However, this is a massive hack. What is the maximum number of fragments? Why do the fragment lengths need expanding out to `size_t`s anyway?

dpgeorge, 2022-09-23:
Fixed by db668742a5986ab1de9c0474ef268c00fb1879e7

closed 2022-09-23 by dpgeorge

stephanelsmith, 2022-09-23:
I was just coming here to mention same thing, MTU at 515.  Will try out latest commit!

stephanelsmith, 2022-09-23:
> Looking at the code, I assume that it is the fragmented buffer reassembly path in `mp_bluetooth_gattc_on_data_available()` that is failing, as the other path doesn't do any heap allocation.
>
> I [...]

ble.config('rxbuf')
I used to have that in my code as well.  I had this set to a fairly large size "global to the entire BLE driver and so handles incoming data for all events, including all characteristics. Increasing this allows better handling of bursty incoming data (for example scan results) and the ability to receive larger characteristic values"
Is this deprecated or not needed?  
https://docs.micropython.org/en/latest/library/bluetooth.html?highlight=irq_scan_done

I might have missed that change.

stephanelsmith, 2022-09-23:
Pulling in https://github.com/micropython/micropython/commit/db668742a5986ab1de9c0474ef268c00fb1879e7
Working for me now.  :beers:

jonathanhogg, 2022-09-23:
> Is this deprecated or not needed? I might have missed that change.

@stephanelsmith: it sets the ring buffer size, but if a port uses the new synchronous BLE events implementation (which recent ESP32 does) then there is no ring buffer and trying to set `rxbuf` will raise an error.

stephanelsmith, 2022-09-23:
> > Is this deprecated or not needed? I might have missed that change.
> 
> @stephanelsmith: it sets the ring buffer size, but if a port uses the new synchronous BLE events implementation (which recent ESP32 does) then there is no ring buffer and trying to set `rxbuf` will raise an error.

:+1: fascinating, thank you.  I will do some digging to learn more.