micropython/micropython #9404: wrong result from b'\xaa \xaa'.count(b'\xaa') (returns 1, expected 2)

rmu75, 2022-09-24:
count() method from bytes sometimes returns wrong counts.

steps to reproduce:
```
MicroPython v1.19.1-dirty on 2022-09-24; linux [GCC 10.3.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> b'\xaa \xaa'.count(b'\xaa')
1
>>>
```
tested on unix and stm32 ports.

expected:
```
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> b'\xaa \xaa'.count(b'\xaa')
2
>>>
```

mattytrentini, 2022-09-25:
It looks like only sequentially occurring substrings are being counted:

```python
MicroPython v1.19.1-60-ge22b7fb4a-dirty on 2022-06-26; linux [GCC 8.3.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> b'\xaa \xaa'.count(b'\xaa') # incorrect
1
>>> b'\xaa\xaa'.count(b'\xaa') # correct
2
>>> b'\xaa\xaa\xaa'.count(b'\xaa') # correct
3
>>> b'\xaa\xaa\xaa \xaa'.count(b'\xaa') # incorrect
3
```

I thought the code that would do this would be in [sequence.c](https://github.com/micropython/micropython/blob/master/py/sequence.c#L208) but that logic _looks_ correct...

jimmo, 2022-09-25:
> I thought the code that would do this would be in [sequence.c](https://github.com/micropython/micropython/blob/master/py/sequence.c#L208) but that logic _looks_ correct...

The problem is that it's using str.count (and trying to interpret this as unicode).

mattytrentini, 2022-09-25:
> The problem is that it's using str.count (and trying to interpret this as unicode).

Ah, right, in [objstr.c](https://github.com/micropython/micropython/blob/master/py/objstr.c#L1747).

jimmo, 2022-09-25:
Everything else in objstr.c that needs to handle both `bytes` (and now `bytearray` and `memoryview`) as well as `str` instances uses `str_index_to_ptr` whereas count is using `utf8_next_char` directly. We need some sort of "get next-char function for type" that `str_count` can use. `str_count` could actually use `str_index_to_ptr` with `index=1` on the previous pointer, but it would be slow because it checks the type every time.

Looking at `str_index_to_ptr` though (the objstrunicode.c version), I think it needs to also check for memoryview.

dpgeorge, 2022-09-25:
Fixed by 9d6f474ea49fd89b7a1a90b830e6014ef70a89b7

closed 2022-09-25 by dpgeorge