micropython/micropython #9405: esp32c3-usb-20220618-v1.19.1.bin can not boot

QGB, 2022-09-25:
It print:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6dc
load:0x403d0000,len:0x28ac
entry 0x403ce000

```
flash commands:
```
In [135]: !esptool --chip esp32c3 --port COM16 erase_flash
     ...:
     ...: !esptool --chip esp32c3 -p COM16 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x
     ...: 0 C:\test\ESP32\esp32c3\MicroPython=esp32c3-usb-20220618-v1.19.1.bin
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.2s
Hard resetting via RTS pin...
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0015dfff...
Compressed 1431808 bytes to 868690...
Wrote 1431808 bytes (868690 compressed) at 0x00000000 in 22.6 seconds (effective 507.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
but `esp32c3-20220117-micropython-v1.18.bin` works well

QGB, 2022-09-25:
the latest `esp32c3-usb-20220829-unstable-v1.19.1-339-g9a826e0f2.bin` also can not boot

dpgeorge, 2022-09-25:
Please try using the non-USB firmware, ie just `esp32c3-2022...`.

QGB, 2022-09-25:
> Please try using the non-USB firmware, ie just `esp32c3-2022...`.

where is the latest non-USB firmware download url

dpgeorge, 2022-09-25:
Find the firmware at: https://micropython.org/download/esp32c3/

QGB, 2022-09-25:
[esp32c3-20220618-v1.19.1.bin](https://micropython.org/resources/firmware/esp32c3-20220618-v1.19.1.bin) is a non-USB firmware ?

QGB, 2022-09-25:
[esp32c3-20220618-v1.19.1.bin](https://micropython.org/resources/firmware/esp32c3-20220618-v1.19.1.bin) works well, thanks @dpgeorge

QGB, 2022-09-25:
```
bin_path : C:/test/ESP32/esp32c3/micropython=esp32c3-20220829-unstable-v1.19.1-339-g9a826e0f2.bin
2022-09-26__01.09.03__.872 ============================================
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.8s
Hard resetting via RTS pin...
2022-09-26__01.09.16__.712 ============================================
esptool.py v3.2
Serial port COM16
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 60:55:f9:72:d7:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00160fff...
Compressed 1441952 bytes to 875555...
Wrote 1441952 bytes (875555 compressed) at 0x00000000 in 22.6 seconds (effective 509.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
2022-09-26__01.09.45__.246 ============================================
Connected to esp32
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
```

closed 2022-09-25 by dpgeorge