micropython/micropython #9410: esp32: socket.bind() fails with a strange error.

robert-hh, 2022-09-25:
Firmware version:
`MicroPython v1.19.1-443-g30e50ab19 on 2022-09-25; ESP32 module with ESP32` or
`MicroPython v1.19.1-443-g30e50ab19 on 2022-09-25; ESP32 module (spiram) with ESP32`
esp-idf version 4.4

This little script:
```py
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('0.0.0.0', 12345))
```
Causes the error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bindfail.py", line 4, in <module>
TypeError: can't convert 'int' object to str implicitly
```
But only if it is executed as frozen bytecode. Importing it as .py or .mpy file from the file system succeeds. Changing the last line to:
```
s.bind(('0.0.0.0', int(12345)))
```
or using a symbol (but not a const) for the port number works as well.

dpgeorge, 2022-09-25:
Thanks for the report.

When frozen, that tuple used as the argument to bind turns in to:
```c
static const mp_obj_int_t const_obj__boot_0_1 = {{&mp_type_int}, {.neg=0, .fixed_dig=1, .alloc=1, .len=1, .dig=(uint16_t*)(const uint16_t[]){0x3039}}};
static const mp_rom_obj_tuple_t const_obj__boot_0 = {{&mp_type_tuple}, 2, {
    MP_ROM_QSTR(MP_QSTR_0_dot_0_dot_0_dot_0),
    MP_ROM_PTR(&const_obj__boot_0_1),
}};
```
You can see that the port number is encoded as a big int.

The code for bind that extracts the port number assumes it's a small int:
```c
static int _socket_getaddrinfo2(const mp_obj_t host, const mp_obj_t portx, struct addrinfo **resp) {
    const struct addrinfo hints = {
        .ai_family = AF_INET,
        .ai_socktype = SOCK_STREAM,
    };

    mp_obj_t port = portx;
    if (mp_obj_is_small_int(port)) {
        ...
```
So the fix is to change that `mp_obj_is_small_int()` to `mp_obj_is_integer()`.

Also, it should probable be frozen as a small int, so that would be a good thing to improve as well.

robert-hh, 2022-09-25:
Thank you for the swift reply. It's always impressive to see how you know every little detail and dependency of MicroPython.
The change from `mp_obj_is_small_int()` to `mp_obj_is_integer()` works. I can make a PR for that, if you like.
Changing the frozen bytecode content however is definitely beyond my experience.

dpgeorge, 2022-09-25:
Thanks for confirming.  I will fix it.

uraich, 2022-09-25:
All I can say: I am impressed!
Many thanks to both of you.

dpgeorge, 2022-09-25:
Fixed by 3abcfb9aecd567c4c74fd3c99703f23c372b7445

closed 2022-09-25 by dpgeorge