micropython/micropython #9429: make BOARD=GENERIC_C3 error

QGB, 2022-09-26:
```
[ 93%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/qgb/github/micropython/extmod/machine_pwm.c.obj
[ 93%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/qgb/github/micropython/extmod/machine_signal.c.obj
[ 93%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/home/qgb/github/micropython/extmod/machine_spi.c.obj
In file included from /home/qgb/github/micropython/extmod/machine_pwm.c:34:
/home/qgb/github/micropython/ports/esp32/machine_pwm.c: In function 'set_freq':
/home/qgb/github/micropython/ports/esp32/machine_pwm.c:247:30: error: 'LEDC_USE_REF_TICK' undeclared (first use in this function); did you mean 'LEDC_USE_RTC8M_CLK'?
             timer->clk_cfg = LEDC_USE_REF_TICK;
                              ^~~~~~~~~~~~~~~~~
                              LEDC_USE_RTC8M_CLK
/home/qgb/github/micropython/ports/esp32/machine_pwm.c:247:30: note: each undeclared identifier is reported only once for each function it appears in
esp-idf/main/CMakeFiles/__idf_main.dir/build.make:2082: recipe for target 'esp-idf/main/CMakeFiles/__idf_main.dir/home/qgb/github/micropython/extmod/machine_pwm.c.obj' failed
make[3]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/home/qgb/github/micropython/extmod/machine_pwm.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/home/qgb/github/micropython/ports/esp32/build-GENERIC_C3'
CMakeFiles/Makefile2:4765: recipe for target 'esp-idf/main/CMakeFiles/__idf_main.dir/all' failed
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[2]: Leaving directory '/home/qgb/github/micropython/ports/esp32/build-GENERIC_C3'
Makefile:135: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/qgb/github/micropython/ports/esp32/build-GENERIC_C3'
make failed with exit code 2
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
Makefile:40: recipe for target 'all' failed
make: *** [all] Error 1
```

robert-hh, 2022-09-26:
Hmm. Strange. I can build it here. Which version of the ESP-IDF do you use?

QGB, 2022-09-26:
```
idf.py --version
ESP-IDF v4.4.2-dirty
```

robert-hh, 2022-09-26:
I'm using IDF vesion 4.4. Maybe that's the difference,

dpgeorge, 2022-09-26:
This was recently fixed, by dd9dcb594c577cb818c336db59a884fd329c3840

IDF 4.4.2 removed `LEDC_USE_REF_TICK` on C3.

closed 2022-09-26 by dpgeorge

robert-hh, 2022-09-26:
The updated v4.4.2 builds fine.