micropython/micropython #9455: Raspberry Pi Pico W network becomes inaccessible when not used for some time

anecdata, 2022-09-28:
Using `MicroPython v1.19.1 on 2022-09-20; Raspberry Pi Pico W with RP2040`, Pico W loses network operation (though seemingly not AP connection since there is typically still an IP address) if no network operations are performed for some time. The maximum non-operating interval between network operations is typically about 5 minutes with Power-Saving on, and at least 10 minutes with Power-Saving off (more testing continues with this mode). The failure mode is typically `OSError: -2`.

If some operation, say a `ping` is done frequently, the network operation is preserved and more complex transactions (e.g., sockets) can be done with much longer intervals.

Code:
```py
import time
import machine
import network
import urequests as requests

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect("ssid", "passw0rd")
while not wlan.isconnected() and wlan.status() >= 0:
    print(f"Connected? {wlan.isconnected()} Status? {wlan.status()}")
    time.sleep(1)
print(wlan.ifconfig()[0])

# increase delay between requests each time
interval_sec = 0
increment_sec = 60
timer_sec = time.time() - interval_sec
while True:
    if time.time() - timer_sec > interval_sec:
        try:
            r = requests.get(TEXT_URL)
            print(f"{interval_sec:>5} {r.status_code} {r.reason.decode()} {r.content}")
            r.close()
            interval_sec += increment_sec
        except OSError as e:
            print(e)
            machine.reset()
        timer_sec = time.time()
```

Output:
```
Connected? False Status? 1
192.168.6.198
    0 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
   60 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
  120 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
  180 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
  240 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
-2
```

Possibly related to power-saving mode (`wlan.config(pm = 0xa11140)`) described in section 3.6.3 [here](https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf).

Documented [here](https://github.com/adafruit/circuitpython/issues/6958) and [here](https://github.com/adafruit/circuitpython/pull/6933#pullrequestreview-1124397727).

jepler, 2022-10-01:
Based on our testing, we've arranged to "default" to power saving mode of 0xa11140 in circuitpython.

anecdata, 2022-10-01:
Also, to clarify the original comment... it does seem to be typically a disconnection from the wifi AP. The presence of an IP address when not connected was confusing.

`wlan.status()` of 3 means:
`#define CYW43_LINK_UP           (3)     ///< Connect to wifi with an IP address`
but the device will still have an IP address even when it has become disconnected (and across restarts). Is this intended, or should that be filed as a separate issue?

peterhinch, 2022-10-02:
I think this depends on the AP because this test runs here with yesterday's daily build with no config changes. Currently I'm seeing
```
 1080 200 OK b'This is a test of Adafruit WiFi!\nIf you can read this, its working :)'
```
I encountered something similar when developing [mqtt_as](https://github.com/peterhinch/micropython-mqtt/blob/master/mqtt_as/README.md). Collaborator @kevinkk525 reported a similar problem on ESP8266: after periods of inactivity it would spontaneously disconnect and the code would go through the reconnect process. This was fixed by `esp.sleep_type(0)`. I never encountered this issue which seemed to hint at an AP connection. I'd be interested to know the mechanism :)

On the basis of the Pico W [documentation](https://datasheets.raspberrypi.com/picow/connecting-to-the-internet-with-pico-w.pdf) (para 3.6.3) on **mqtt_as** I use 0xa11140 (as per @jepler ).

anecdata, 2022-10-02:
I'm interested to know the mechanism too, though I'm not too savvy to the low layer protocols. Other devices stay connected, but it's also not surprising that there would be differences in behavior across different network equipment when tweaking sleep intervals.

maxi07, 2023-10-29:
Is there any update on this? I am still having the same issue

Martvse, 2023-10-29:
I am also experiencing this problem, it's frustrating.

proxict, 2023-11-02:
I'm running into the same problem with the C API. That makes me believe the problem is more low-level and not in the micropython bindings.