micropython/micropython #9475: MicroPython for RP2 Pico-W missing 'ustruct' module

hridpath, 2022-09-30:
I have installed **rp2-pico-w-20220930-unstable-v1.19.1-461-g282401da5.uf2** on a Raspberry Pi PICO-W. I am using a GC9A01 display with  **_russhughs/gc9a01py_**  driver. 

However, it seems that this version of rp2-pico-w.....uf2 does not include the 'ustruct' module?? The display driver code uses 'ustruct' to pack  data.

Can this be added?

dpgeorge, 2022-10-01:
What does `help("modules")` report, when run at the REPL prompt?

jimmo, 2022-10-01:
`struct` (and `ustruct`) are enabled at the "core features" level, and rp2 uses "extra features", so it should definitely be in your firmware (unless something else is going wrong).

peterhinch, 2022-10-01:
It works here:
```python
MPY: soft reboot
MicroPython v1.19.1 on 2022-09-30; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import ustruct
>>> dir(ustruct)
['__class__', '__name__', '__dict__', 'calcsize', 'pack', 'pack_into', 'unpack', 'unpack_from']
>>> 
```

hridpath, 2022-10-02:
I nuked my RP2040 and reloaded the rp2-pico-20220618-v1.19.1.uf2 version and it works as you described.

Thanks to all of you that do this work.

closed 2022-10-02 by hridpath