micropython/micropython #9489: py: Calls like  mp_obj_is_type(args[0], &mp_type_int) fail.

robert-hh, 2022-10-02:
Calls like `mp_obj_is_type(args[0].u_obj, &mp_type_int)` fail with a compiler error:
```
/home/robert/Downloads/MicroPython/micropython/py/misc.h:54:50: error: size of unnamed array is negative
 #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
                                                  ^
/home/robert/Downloads/MicroPython/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
 #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
                                        ^~~~~~~~~~~~~~~~
/home/robert/Downloads/MicroPython/micropython/py/obj.h:904:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_int), assert((t) != &mp_type_int),           \
     ^~~~~~~~~~~~~~~~~~~~~~~~
/home/robert/Downloads/MicroPython/micropython/py/obj.h:909:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
 #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
```
For the test, I modified the freq() method of modmachine in the following way, just to have a simple error. Initially the error popped up while compiling PR 8766:
```
STATIC mp_obj_t machine_freq(size_t n_args, const mp_obj_t *args) {
    if (n_args == 0) {
        return MP_OBJ_NEW_SMALL_INT(mp_hal_get_cpu_freq());
    } else {
        if (mp_obj_is_type(args[0], &mp_type_int)) {
            mp_int_t freq = mp_obj_get_int(args[0]);
            if (!set_sys_clock_khz(freq / 1000, false)) {
                mp_raise_ValueError(MP_ERROR_TEXT("cannot change frequency"));
            }
            #if MICROPY_HW_ENABLE_UART_REPL
            setup_default_uart();
            mp_uart_init();
            #endif
        }
        return mp_const_none;
    }
}
STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(machine_freq_obj, 0, 1, machine_freq);
```
Using another type like mp_type_float does not create the error.

jimmo, 2022-10-02:
See https://github.com/micropython/micropython/pull/5559

It's to prevent an easy trap of missing that the obj is a short int.

Either use mp_obj_is_int (preferred) or mp_obj_is_exact_type if you know that it isn't a short int.

(Similar story for strings, which may be qstrs)

robert-hh, 2022-10-02:
When was that changed? It's not about my code, but it used to work a few months ago.

jimmo, 2022-10-02:
It was merged 18 July this year.

robert-hh, 2022-10-02:
Thanks. Works with mp_obj_is_int().

closed 2022-10-02 by robert-hh