micropython/micropython #9496: tools/pyboard.py tries to "import uos" on repl client -- there is no uos.

n1kdo, 2022-10-03:
Kinda like #9018 and kinda like #7499.

tools/pyboard.py tries to "import uos" for filesystem related operations in six places. 

There is no "uos", but there is "os".  I modified my local copy of pyboard.py to "import os" in these six places and that works.

I'm happy to make a PR for this. Please advise.

Thanks!

robert-hh, 2022-10-03:
Which firmware version is loaded to the PyBoard. With the board I have here I can import uos. What is the output of

`help("modules")`

jimmo, 2022-10-03:
@n1kdo for some more context here: On "official" MicroPython firmware (i.e. this repo), the `os` module is actually called `uos`, and `os` is an alias. This allows the module to be extended from Python via `os.py` on the filesystem. Because 99% of users don't care about this, the docs and most examples just refer to it as `os`.

However, it's important for `pyboard.py` to use `uos` because it will bypass any possible extensions that are installed on the filesystem. Also, there are some ports that don't enable the aliasing, in which case you _must_ `import uos`.

Personally I think the u-prefix adds more confusion than the small benefit it gains. For a lot more background on this (including a proposal to remove the u-prefix everywhere), see
https://github.com/micropython/micropython/issues/9018
https://github.com/micropython/micropython/pull/9069

Other forks of MicroPython (notably CircuitPython) don't do this -- the `os` module is called `os`. So that would be my guess for what's happening in your case?

It might be worth considering making `pyboard.py` auto-detect... i.e. change the code to always do `try: import uos as os; except: import os` in all cases, and then perhaps it would "just work" on CircuitPython too!

dpgeorge, 2022-10-03:
> It might be worth considering making `pyboard.py` auto-detect...

If we go down this path, I think it's better to auto-detect the board capabilities once at start up (eg what modules it has available) and then use those from then on.

n1kdo, 2022-10-04:
```
MicroPython v1.19.1 on 2022-07-29; Raspberry Pi Pico W with RP2040

Type "help()" for more information.
>>> help("modules")
__main__          gc                uasyncio/lock     upip
_boot             lwip              uasyncio/stream   upip_utarfile
_boot_fat         math              ubinascii         urandom
_onewire          micropython       ucollections      ure
_rp2              neopixel          ucryptolib        urequests
_thread           network           uctypes           uselect
_uasyncio         onewire           uerrno            usocket
_webrepl          rp2               uhashlib          ussl
builtins          uarray            uheapq            ustruct
cmath             uasyncio/__init__ uio               usys
dht               uasyncio/core     ujson             utime
ds18x20           uasyncio/event    umachine          uwebsocket
framebuf          uasyncio/funcs    uos               uzlib
Plus any modules on the filesystem
>>> 
```

I got the latest Pico-W build (```[v1.19.1-468-g0e35c4de9 (2022-10-04) .uf2](https://micropython.org/resources/firmware/rp2-pico-w-20221004-unstable-v1.19.1-468-g0e35c4de9.uf2```) and the latest pyboard.py (Jimmo's 7705b9b) I cannot reproduce this. Sorry about the confusion.

closed 2022-10-04 by n1kdo