micropython/micropython #9500: RP2 uasync TCP server calls handler with no request (1.19.1.468 (4.10.2022))

msetina, 2022-10-04:
micropython:
Using a uasync start_server to get TCP server I get a random request with
reader.readline returning b''
The problem is that there is an allocated memory after that of most of memory (alloc:134336, free:31680) and does not get reclamed even with gc.collect

msetina, 2022-10-04:
Memory problem was located and mitigated. What reamains is a call that has no data behind.

dpgeorge, 2022-10-05:
Can you please share the Python code that has this problem?

msetina, 2022-10-05:
It should work with any sample:
import network
import time

import uasyncio as asyncio

ssid = "<SSID>"
password = "<password>"

wlan = network.WLAN(network.STA_IF)

def connect_to_network():
    wlan.active(True)
    #wlan.config(pm = 0xa11140)  # Disable power-save mode
    wlan.connect(ssid, password)

    max_wait = 10
    while max_wait > 0:
        if wlan.status() < 0 or wlan.status() >= 3:
            break
        max_wait -= 1
        print('waiting for connection...')
        time.sleep(1)

    if wlan.status() != 3:
        raise RuntimeError('network connection failed')
    else:
        print('connected')
        status = wlan.ifconfig()
        print('ip = ' + status[0])

async def serve_client(reader, writer):
    print("Client connected")
    request_line = await reader.readline()
    print("Request:", request_line)
    # We are not interested in HTTP request headers, skip them
    while await reader.readline() != b"\r\n":
        pass

    request = str(request_line)
    print(request)
        
    response = ""
    writer.write('HTTP/1.0 200 OK\r\nContent-type: text/html\r\n\r\n')
    writer.write(response)

    await writer.drain()
    await writer.wait_closed()
    print("Client disconnected: {0}".format(request_line))

async def main():
    print('Connecting to Network...')
    connect_to_network()

    print('Setting up webserver...')
    asyncio.create_task(asyncio.start_server(serve_client, '<you IP>', 80))
    while True:        
        print("heartbeat")
        await asyncio.sleep(0.25)
        print("AAA")
        await asyncio.sleep(5)
        
try:
    asyncio.run(main())
finally:
    asyncio.new_event_loop()

The case is that First request after device start from Chromium(Version 105.0.5195.125 (Official Build) snap (64-bit)) will see 3 "Client connected" instead of 2.

Connecting to Network...
waiting for connection...
waiting for connection...
waiting for connection...
connected
ip = <your IP>
Setting up webserver...
heartbeat
AAA
heartbeat
AAA
heartbeat
AAA
Client connected
Request: b'GET / HTTP/1.1\r\n'
b'GET / HTTP/1.1\r\n'
Client disconnected: b'GET / HTTP/1.1\r\n'
Client connected
Client connected
Request: b'GET /favicon.ico HTTP/1.1\r\n'
b'GET /favicon.ico HTTP/1.1\r\n'
Client disconnected: b'GET /favicon.ico HTTP/1.1\r\n'

But waiting for sime time the remaing request/call wakes up and the continuation wll result in:
AAA
heartbeat
AAA
Request: b''
heartbeat
AAA
heartbeat

The code is just a modified example from internet.

dpgeorge, 2022-10-05:
I tried your example and it works OK for me with Firefox and Chromium (Version 104.0.5112.101 (Official Build) Arch Linux (64-bit)).  Both make only 2 requests (one for / and one for /favicon.ico).  There is no 3rd request.

So I'm not sure what happens in your case.  Maybe try running the same code (without the `connect_to_network()` function) on normal CPython and see what happens there.

`readline()` is allowed to return `b""`.  That means the connection was closed without sending any data.

msetina, 2022-10-06:
I understand. I will add firefox to test regime and try on some Linux distro.
But I have to repeat that it is sporatic issue. I could get it to be repeatable after hard reset, but not on soft reset.

I understand b'' is a valid result from readline()

msetina, 2022-10-10:
Python 3.10.6  with the same Chrome does not do the extra call.

dpgeorge, 2022-10-11:
To make progress here you'd need to trace the network packets that go back and forth.  This can be done via one or more of the following methods:
1. Opening the debugging panel on Chrome via F12, then clicking on the "Network" tab.  Then watch the requests as you access the page.
2. Enable tracing on the PICO_W via `wlan.config(trace=15)` (before starting the server) and capturing the output.
3. Using Wireshark on your PC to capture packets.

If you can report the output of tracing, that could be helpful to understand the problem.

msetina, 2022-11-20:
Here is a trace:
[   61050] ETHRX len=74 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49890 dstport=8080 seqnum=465959595 acknum=0 dataoff=10 flags=2 winsz=64240 opts= 02 04 05 b4 04 02 08 0a b0 d9 4a fb 00 00 00 00 01 03 03 07 data=
[   61052] ETHTX len=42 dst=ff:ff:ff:ff:ff:ff src=28:cd:c1:00:89:69 type=ARP data= 00 01 08 00 06 04 00 01 28 cd c1 00 89 69 c0 a8 2c 18 00 00 00 00 00 00 c0 a8 2c 10
[   61053] ETHRX len=74 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49896 dstport=8080 seqnum=664737748 acknum=0 dataoff=10 flags=2 winsz=64240 opts= 02 04 05 b4 04 02 08 0a b0 d9 4a ff 00 00 00 00 01 03 03 07 data=
[   61151] ETHRX len=74 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49908 dstport=8080 seqnum=1728798689 acknum=0 dataoff=10 flags=2 winsz=64240 opts= 02 04 05 b4 04 02 08 0a b0 d9 4b f6 00 00 00 00 01 03 03 07 data=
[   61253] ETHRX len=42 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=ARP data= 00 01 08 00 06 04 00 02 a0 a8 cd c2 31 84 c0 a8 2c 10 28 cd c1 00 89 69 c0 a8 2c 18
[   61254] ETHTX len=58 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49908 seqnum=6562 acknum=1728798690 dataoff=6 flags=12 winsz=6400 opts= 02 04 03 20 data=

[   61271] ETHRX len=54 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49908 dstport=8080 seqnum=1728798690 acknum=6563 dataoff=5 flags=10 winsz=64240 data=
[   61273] ETHTX len=54 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49908 seqnum=6563 acknum=1728799160 dataoff=5 flags=10 winsz=5930 data=
Client connected
Resource /. Method GET
[   62073] ETHRX len=74 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49896 dstport=8080 seqnum=664737748 acknum=0 dataoff=10 flags=2 winsz=64240 opts= 02 04 05 b4 04 02 08 0a b0 d9 4e e9 00 00 00 00 01 03 03 07 data=
[   62075] ETHTX len=58 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49896 seqnum=6554 acknum=664737749 dataoff=6 flags=12 winsz=6400 opts= 02 04 03 20 data=
[   62076] ETHRX len=74 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49890 dstport=8080 seqnum=465959595 acknum=0 dataoff=10 flags=2 winsz=64240 opts= 02 04 05 b4 04 02 08 0a b0 d9 4e e9 00 00 00 00 01 03 03 07 data=
[   62078] ETHTX len=58 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49890 seqnum=6546 acknum=465959596 dataoff=6 flags=12 winsz=6400 opts= 02 04 03 20 data=
[   62176] ETHRX len=54 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49896 dstport=8080 seqnum=664737749 acknum=6555 dataoff=5 flags=10 winsz=64240 data=
[   62177] ETHRX len=54 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49890 dstport=8080 seqnum=465959596 acknum=6547 dataoff=5 flags=10 winsz=64240 data=
Client connected
Client connected
No content in set time
[   62347] ETHTX len=98 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49908 seqnum=6563 acknum=1728799160 dataoff=5 flags=19 winsz=5930 data= 48 54 54 50 2f 31 2e 30 20 32 30 30 20 4f 4b 0d 0a 43 6f 6e 74 65 6e 74 2d 74 79 70 65 3a 20 74 65 78 74 2f 68 74 6d 6c 0d 0a 0d 0a
Client disconnected from /
[   62362] ETHRX len=54 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49908 dstport=8080 seqnum=1728799160 acknum=6608 dataoff=5 flags=11 winsz=64195 data=
[   62364] ETHTX len=54 dst=a0:a8:cd:c2:31:84 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.44.16 TCP srcport=8080 dstport=49908 seqnum=6608 acknum=1728799161 dataoff=5 flags=10 winsz=5929 data=
[   62412] ETHRX len=445 dst=28:cd:c1:00:89:69 src=a0:a8:cd:c2:31:84 type=IPv4 srcip=192.168.44.16 dstip=192.168.44.24 TCP srcport=49896 dstport=8080 seqnum=664737749 acknum=6555 dataoff=5 flags=18 winsz=64240 data= 47 45 54 20 2f 66 61 76 69 63 6f 6e 2e 69 63 6f 20 48 54 54 50 2f 31 2e 31 0d 0a 48 6f 73 74 3a 20 31 39 32 2e 31 36 38 2e 34 34 2e 32 34 3a 38 30 38 30 0d 0a 43 6f 6e 6e 65 63 74 69 6f 6e 3a 20 6b 65 65 70 2d 61 6c 69 76 65 0d 0a 55 73 65 72 2d 41 67 65 6e 74 3a 20 4d 6f 7a 69 6c 6c 61 2f 35 2e 30 20 28 58 31 31 3b 20 4c 69 6e 75 78 20 78 38 36 5f 36 34 29 20 41 70 70 6c 65 57 65 62 4b 69 74 2f 35 33 37 2e 33 36 20 28 4b 48 54 4d 4c 2c 20 6c 69 6b 65 20 47 65 63 6b 6f 29 20 43 68 72 6f 6d 65 2f 31 30 36 2e 30 2e 30 2e 30 20 53 61 66 61 72 69 2f 35 33 37 2e 33 36 0d 0a 41 63 63 65 70 74 3a 20 69 6d 61 67 65 2f 61 76 69 66 2c 69 6d 61 67 65 2f 77 65 62 70 2c 69 6d 61 67 65 2f 61 70 6e 67 2c 69 6d 61 67 65 2f 73 76 67 2b 78 6d 6c 2c 69 6d 61 67 65 2f 2a 2c 2a 2f 2a 3b 71 3d 30 2e 38 0d 0a 52 65 66 65 72 65 72 3a 20 68 74 74 70 3a 2f 2f 31 39 32 2e 31 36 38 2e 34 34 2e 32 34 3a 38 30 38 30 2f 0d 0a 41 63 63 65 70 74 2d 45 6e 63 6f 64 69 6e 67 3a 20 67 7a 69 70 2c 20 64 65 66 6c 61 74 65 0d 0a 41 63 63 65 70 74 2d 4c 61 6e 67 75 61 67 65 3a 20 73 6c 2d 53 49 2c 73 6c 3 [... cut ...]
[   64466] ETHTX len=54 dst=00:0d:b9:51:a6:97 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.1.109 TCP srcport=59611 dstport=80 seqnum=6671 acknum=3758711701 dataoff=5 flags=10 winsz=5891 data=
[   64513] ETHTX len=54 dst=00:0d:b9:51:a6:97 src=28:cd:c1:00:89:69 type=IPv4 srcip=192.168.44.24 dstip=192.168.1.109 TCP srcport=59611 dstport=80 seqnum=6671 acknum=3758711701 dataoff=5 flags=11 winsz=5891 data=
[   64518] ETHRX len=56 dst=28:cd:c1:00:89:69 src=00:0d:b9:51:a6:97 type=IPv4 srcip=192.168.1.109 dstip=192.168.44.24 TCP srcport=80 dstport=59611 seqnum=3758711701 acknum=6672 dataoff=5 flags=10 winsz=5840 data=
No return!
No return!