micropython/micropython #9505: IPv4 address can persist well after wifi AP connection is lost

anecdata, 2022-10-04:
`MicroPython v1.19.1 on 2022-09-20; Raspberry Pi Pico W with RP2040`

As noted in a comment in [#9455](https://github.com/micropython/micropython/issues/9455#issuecomment-1264414826):

wlan.status() of 3 means:
`#define CYW43_LINK_UP (3) Connect to wifi with an IP address`
as opposed to:
`#define | CYW43_LINK_NOIP (2) Connected to wifi, but no IP address`

But the device will still have an IP address even when it has become disconnected from the wifi Access Point (at least when it disconnects through the mechanism in 9455). Is this an intended feature, emergent feature, or bug? The IP address also sometimes returns `0.0.0.0`, which I suspect is the `wlan.status() == 2` case, and is at least is easy to check for.

From #8994:

> The idea with soft reset is that it resets Python state but not necessarily hardware state. Eg soft reset won't put GPIO back to a default state, or disable USB. Any hardware that's left untouched after a soft reset should still be in a valid state, and if code expects to run after a soft reset then it must be aware that hardware peripherals can be in a variety of states when MicroPython code starts running.

But this doesn't seem to fall within the realm of hardware state, since the device shows an IP address at times when it doesn't actually have a connection to an AP and, particularly in the DHCP case, doesn't have a _valid_ IP address.

jimmo, 2022-10-05:
Yes I think this is a bug. I noticed as well that the IP address persists even when you disable the wifi interface.

I am working on this at the moment as part of a larger fix to the Pico W (and PYBD) WiFi driver to make the API&behavior more consistent with the other WiFi-enabled ports (ESP32, ESP8266).

kentindell, 2022-10-10:
> Yes I think this is a bug. I noticed as well that the IP address persists even when you disable the wifi interface.
>
> I am working on this at the moment as part of a larger fix to the Pico W (and [...]

Is there a plan for when the next formal release will be? I like to drop our custom firmware based on release points and the last formal release doesn't include Pico W support.