micropython/micropython #9525: Invalid string format (single dot) not raising an exception

Neradoc, 2022-10-05:
The wrong format specifier `"{:.}"` does not raise an error, when it does in C python.
```py
MicroPython v1.19.1 on 2022-10-05; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> f"{5:.}"
'5'
>>> f"{'hi':.}"
'hi'
>>> f"{True:.}"
'1'
```
Expected:
```py
Python 3.9.6 (default, Aug 11 2021, 01:29:00) 
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> f"{5:.}"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Format specifier missing precision
```

jimmo, 2022-10-05:
Thanks @Neradoc !

The fix is very small (+8 bytes) but I don't really have a good feeling for how worthwhile it is... it doesn't seem too serious to allow this. OTOH, CPython compatibility is always good.

Anyway, https://github.com/micropython/micropython/pull/9526 fixes it and adds tests.

Neradoc, 2022-10-05:
Thanks for the quick response.
Yeah I don't know how important it is, "won't fix" is a valid answer for me if the size difference is not worth it.
It's not like it's something that people would chose to use intentionally just because it's possible.

dpgeorge, 2023-01-20:
We have decided not to fix this.

closed 2023-01-20 by dpgeorge