micropython/micropython #9533: webrepl difficult or impossible to connect to if cpu is busy

PsuFan, 2022-10-06:
I find the webrepl difficult or impossible to connect to if you dont have a sleep in your loop. Is there a way to do a webrepl.accept_conn in your loop instead of sleeping?

PsuFan, 2022-10-11:
Esp32, I believe 8266 as well

PsuFan, 2022-10-11:
Ugh the only GitHub options on mobile for reply were close and comment….

PsuFan, 2022-10-14:
anything on this? You can test this with while True: pass; It does not allow you to connect to webrepl.

daniel-da, 2022-11-16:
Have you tried to run your code in a _thread?
So it will leave the main thread available for your webrepl connection

PsuFan, 2022-11-16:
no I haven't, can I run webrepl in a _thread?

daniel-da, 2022-11-16:
I was talking more about your process or task, like the while True: pass

daniel-da, 2022-11-16:
this is the snippet I used to test on an ESP32, not sure this will work on an ESP8266:

```python
import _thread

async def loadSystem():
    while True:
        pass

_thread.start_new_thread(loadSystem,())
```

PsuFan, 2022-11-16:
simple enough, I'll have to give it a shot. Thanks but there still should be a call we can make in our own loop to handle webrepl.

daniel-da, 2022-11-16:
I've just call `webrepl.start()` in my `main.py` just after the wifi connection, and not in the loop

daniel-da, 2022-11-16:
```python
import webrepl
import network

sta_if = network.WLAN(network.STA_IF)
if not sta_if.isconnected():
    print('connecting to network...')
    sta_if.active(True)
    sta_if.connect('MyHome_2G', 'aSwSWNkntLSU9ocmllBNoNOGh2lW36V2')
    while not sta_if.isconnected():
        pass
print('network config:', sta_if.ifconfig())
webrepl.start()
```

PsuFan, 2022-11-16:
Yeah try and connect to webrepl while running: while True: pass;