micropython/micropython #9535: SD card file modification date/time wrong--Vfat Raspberry Pi PicoW

Wbillot, 2022-10-06:
* Please search existing issues before raising a new issue. For questions about MicroPython or for help using MicroPython, or any sort of "how do I?" requests, please use the Discussions tab or raise a documentation request instead.

* In your issue, please include a clear and concise description of what the bug is, the expected output, and how to replicate it.

* If this issue involves external hardware, please include links to relevant datasheets and schematics.

* If you are seeing code being executed incorrectly, please provide a minimal example and expected output (e.g. comparison to CPython).

* For build issues, please include full details of your environment, compiler versions, command lines, and build output.

* Please provide as much information as possible about the version of MicroPython you're running, such as:
 - firmware file name
 - git commit hash and port/board
 - version information shown in the REPL (hit Ctrl-B to see the startup message)

* Remove all placeholder text above before submitting.

scruss, 2022-10-07:
some background to this for the dev team, if needed: [Pico W writes to SD card have wrong file date and time - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=341054)

robert-hh, 2022-10-07:
That can be easily fixed in the file fatfs_port.c. The year calculation there seems odd. If no on else raises the hand, I will look into it in the next days.

robert-hh, 2022-10-07:
I made a PR for a fix. Actually the year was wrong by 2000 years, but in the directory entry, the year field is 7 bit long. So the offset ended as 80 years. 2000 % 128 == 80.

peterhinch, 2022-10-08:
On this topic I am finding that copying files to on-board Flash also produces incorrect timestamps, offset by one hour. This occurs whether I use `rshell` or `mpremote` to do the copying. If my PC time is 10:00, the timestamp of the file is 11:00.

This causes `rsync` on `mpremote` to fail to update newly changed files.

I did wonder if it's a DST problem but UK time is currently UTC + 1 so presumably the error would be in the opposite direction.

peterhinch, 2022-10-12:
@robert-hh Have you encountered the above?

robert-hh, 2022-10-12:
The modification date fix was merged yesterday as 92d911803. It fixes the year, and the time written is the one set with RTC module. When I read it at my PC, I get +2 hours. I am at UTC+2. So I assume that the PC takes the file time as UTC time and adds the timezone offset.

scruss, 2022-10-12:
@peterhinch If I create a file on the card with my Linux computer at 2022-10-12 12:57:58 local time, MicroPython's `stat()` returns 2022-10-12 16:57:58, as expected. My time zone is UTC-4, and the RTC on my Pico is set to local time.

@robert-hh Thank you for fixing this!

peterhinch, 2022-10-13:
@robert-hh I've figured this out. The fault seems to be in `rshell`: it is setting the Pico RTC at +1hr relative to the PC's clock. I'll look at the code and see if I can figure out why this is happening.